OrcaSlicer/OrcaSlicer #4671: Slicing a file with fuzzy skin causes an EXC_BAD_ACCESS (SIGSEGV) crash.

fenhelix, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

macOS

### OS Version

macOS 14.3.1 (23D60)

### Additional system information

  Model Name:	Mac Studio
  Model Identifier:	Mac14,14
  Model Number:	G180CLL/A
  Chip:	Apple M2 Ultra
  Total Number of Cores:	24 (16 performance and 8 efficiency)
  Memory:	64 GB
  System Firmware Version:	10151.81.1
  OS Loader Version:	10151.81.1

Apple M2 Ultra GPU:

  Chipset Model:	Apple M2 Ultra
  Type:	GPU
  Bus:	Built-In
  Total Number of Cores:	76
  Vendor:	Apple (0x106b)
  Metal Support:	Metal 3
  Displays:
Odyssey G93SC:
  Resolution:	5120 x 1440
  UI Looks like:	5120 x 1440 @ 240.00Hz
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported

### Printer

X1C

### How to reproduce

Open attached file, and hit slice. 

### Actual results

Crash and reopen system dialog.

### Expected results

The model should slice and not crash Orca

### Project file & Debug log uploads

[Box 4.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14734121/Box.4.3mf.zip)
[debug_Sat_Mar_23_21_04_25_84990.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14734151/debug_Sat_Mar_23_21_04_25_84990.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Maybe related to #4526 

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [84990]

VM Region Info: 0 is not in any region.  Bytes before following region: 4301684736
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100668000-10424c000    [ 59.9M] r-x/r-x SM=COW  ...OS/OrcaSlicer

Thread 29 Crashed:: slic3r_tbb_19
0   OrcaSlicer                    	       0x102b8a3a8 Slic3r::traverse_loops(Slic3r::PerimeterGenerator const&, std::__1::vector<Slic3r::PerimeterGeneratorLoop, std::__1::allocator<Slic3r::PerimeterGeneratorLoop>> const&, std::__1::vector<Slic3r::ThickPolyline, std::__1::allocator<Slic3r::ThickPolyline>>&, bool&, bool&) + 6496
1   OrcaSlicer                    	       0x102b8a370 Slic3r::traverse_loops(Slic3r::PerimeterGenerator const&, std::__1::vector<Slic3r::PerimeterGeneratorLoop, std::__1::allocator<Slic3r::PerimeterGeneratorLoop>> const&, std::__1::vector<Slic3r::ThickPolyline, std::__1::allocator<Slic3r::ThickPolyline>>&, bool&, bool&) + 6440
2   OrcaSlicer                    	       0x102b82c94 Slic3r::PerimeterGenerator::process_classic() + 6740
3   OrcaSlicer                    	       0x102a8d014 Slic3r::LayerRegion::make_perimeters(Slic3r::SurfaceCollection const&, Slic3r::SurfaceCollection*, std::__1::vector<Slic3r::ExPolygon, std::__1::allocator<Slic3r::ExPolygon>>*) + 628
4   OrcaSlicer                    	       0x102a8987c Slic3r::Layer::make_perimeters() + 4608
5   OrcaSlicer                    	       0x102d6a584 tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, Slic3r::PrintObject::make_perimeters()::$_3, tbb::detail::d1::auto_partitioner const>::execute(tbb::detail::d1::execution_data&) + 824
6   OrcaSlicer                    	       0x1051e456c tbb::detail::d1::task* tbb::detail::r1::task_dispatcher::local_wait_for_all<false, tbb::detail::r1::outermost_worker_waiter>(tbb::detail::d1::task*, tbb::detail::r1::outermost_worker_waiter&) + 932
7   OrcaSlicer                    	       0x1051e165c tbb::detail::r1::arena::process(tbb::detail::r1::thread_data&) + 544
8   OrcaSlicer                    	       0x1051ea0bc tbb::detail::r1::market::process(rml::job&) + 52
9   OrcaSlicer                    	       0x1051ec8d4 tbb::detail::r1::rml::private_worker::run() + 224
10  OrcaSlicer                    	       0x1051ec7e8 tbb::detail::r1::rml::private_worker::thread_routine(void*) + 12
11  libsystem_pthread.dylib       	       0x189eaa034 _pthread_start + 136
12  libsystem_pthread.dylib       	       0x189ea4e3c thread_start + 8

SoftFever, 2024-03-24:
It should be fixed in 2.0.0 RC release.
Can you give it a try?
https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.0.0-rc

fenhelix, 2024-03-24:
> It should be fixed in 2.0.0 RC release. Can you give it a try? https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.0.0-rc

Confirmed fixed in 2.0.0 RC ✅

closed 2024-03-24 by fenhelix