OrcaSlicer/OrcaSlicer #4672: Improve "Further reduce solid infill on walls" logic [BETA]

HakunMatat4, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-dev

### Operating System (OS)

Linux

### OS Version

Mint Cinnamon 

### Printer

Voron Trident 

### How to reproduce

This is continuation of: https://github.com/SoftFever/OrcaSlicer/pull/3235
* Enable "Further reduce solid infill on walls"

### Actual results

I've noticed this in some prints but totally forgot about it.
This time tho, I had to understand what is wrong.
Left had the feature enabled while the right one has it disabled, via are present.

[image]

### Expected results

The logic should in theory, detect when the overhang is to great and not apply changes.
So that allows you to save filament when it's necessary without compromising quality in the areas that do require wall support to avoid void.

### Project file & Debug log uploads

Any print with round tops where gap infill support would usually be added.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-03-24:
Hey!! In 2.0 there are 3 options now - which one did you pick?

HakunMatat4, 2024-03-24:
> Hey!! In 2.0 there are 3 options now - which one did you pick?

Hey, I am not running 2.0.0 Beta just yet, too many changes which is good but also means possible new bugs and I just don't wanna spend time with dramas haha after spending a long time down upgrading my toolhead.

I am running SoftFever fix for the max vol speed: https://github.com/SoftFever/OrcaSlicer/pull/4398

closed 2024-06-30 by github-actions[bot]