OrcaSlicer/OrcaSlicer #4679: I lost my configs and profiles after the update

fylhtq7779, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.8.1,1.9.1,2.0rc

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Kobra 2 neo (klipper)

### How to reproduce

Update OrcaSlicer

### Actual results

After the update, my profiles didn't show up in the new version
They are still in %appdata%, but I can't import them in any way
I tried version 2.0rc, tried 1.91
Then tried downloading the version that was 1.8.1 and nothing changed.
I don't know what to do

### Expected results

I think profiles should be automatically updated

### Project file & Debug log uploads

[debug_Sun_Mar_24_15_26_59_18704.log](https://github.com/SoftFever/OrcaSlicer/files/14735436/debug_Sun_Mar_24_15_26_59_18704.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

Rickthebig, 2024-03-25:
In 2.0 Beta, you could lose the profiles by just restarting Orca. without any update! Happens ocassionally.

DVSVIDEO, 2024-03-25:
The same thing happened to me with the V2.0.0 beta.
I lost my printer settings, which is a pain because it's not a factory printer, but a printer I built and so I don't have a profile for it.
I also lost all the material profiles.
The previously saved profiles refused to load.
That's the price you pay for newer features.
I didn't mind....uploaded it again.

closed 2024-07-01 by github-actions[bot]