OrcaSlicer/OrcaSlicer #4682: Prominent Hull Line artifact on 3DBenchy, if "Ensure vertical shell thickness" is not "None"

Thinkersbluff, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-RC

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Processor	Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz   1.80 GHz
Installed RAM	20.0 GB (19.8 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Creality CR6-SE

### How to reproduce

1. Load the attached Project .3mf file.   

2. Slice 3DBenchy.stl with "Ensure vertical shell thickness" set to any value except "None".
3. Verify Preview looks fine.
4. Print model with layer height of 0.2mm
5. Observe that at layer 46, the hull is drawn with a prominent offset from the previous layer, as if the slope of the hull up to that point has been too far inboard (see lower hull in photo).
[image]

6. Repeat above, but set "Ensure vertical shell thickness" to "None".
7. Observe that hull prints correctly, without that artifact.

### Actual results

As described and shown above, the sliced 3DBenchy model looks fine in the Preview screen, but prints with a significant hull line artifact at layer 46 and subs, if "Ensure vertical shell thickness" is set to any value except "None".

NOTE: This same bug also appears in 2.0.0-beta.  In 2.0.0-beta, the top shell also became only one layer thick, regardless of the top layer settings, when "Ensure vertical shell thickness" was set to "None".  In 2.0.0-RC, that bug has been fixed, but this one remains.

### Expected results

I am not sure how "Ensure vertical shell thickness" is supposed to differ from or modify the effects of the "Top/Bottom shells" settings.  At the moment, I see no difference in the slicer preview, but I definitely see a difference in the printed model, as described above.
If  "Ensure vertical shell thickness" should be set to "None" when printing the 3DBenchy.stl, that is not clear from the Help files or from the Preview.
If it should NOT be set to "None", then it should not leave that artifact, which appears to mean that it should not be modifying the hull slope.

### Project file & Debug log uploads

Project File: [2-0-0-RC_3DBenchy_vs_VerticalShellThickness.zip](https://github.com/SoftFever/OrcaSlicer/files/14735854/2-0-0-RC_3DBenchy_vs_VerticalShellThickness.zip)
Debug Log: [debug_Fri_Mar_22_22_22_37_23856.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14735892/debug_Fri_Mar_22_22_22_37_23856.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2024-03-29:
Firstly don’t use “none”. It is there for speed benchy type prints and it will mess up most of your regular prints. Instead pick one of the other three options and see what it does to your infill.

what you’re see is expected. It’s because of material shrinking when cooling. Less walls (due to less solid infill) results in higher shrinkage in those areas.

Thinkersbluff, 2024-03-29:
I used all four settings.  Only "None" printed correctly.
Cura has no problem printing the 3D Benchy.
It should not be "norrmal" to generate failed prints, with the default slicer settings.
I consider this a bug, not a feature.

igiannakas, 2024-03-29:
https://help.prusa3d.com/article/the-benchy-hull-line_124745

closed 2024-07-05 by github-actions[bot]