OrcaSlicer/OrcaSlicer #4685: Pressure Advance - consistent line size - Bambu Lab P1S

Yorshka-Vermilion, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

10

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to Prepare tab
2. Click on Calibration
3. Pressure Advance
4. DDE, PA Line, 0, 0.3, 0.005
5. Ok
6. Slice
7. Print

### Actual results

Consistent line size across all k values

### Expected results

Not consistent line sizes

### Project file & Debug log uploads

[Pressure Advance Test.zip](https://github.com/SoftFever/OrcaSlicer/files/14736232/Pressure.Advance.Test.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

I've also encounter it with Tower PA. Same quality on every level. G-code seems ok, M900 codes with different K values are present. 
I have latest P1S firmware installed, i've also run Flow Dynamics test from BambuStudio and it went without problems, every line was different.

vgdh, 2024-03-24:
can pursue advance be disabled  in printer settings?

Yorshka-Vermilion, 2024-03-25:
> can pursue advance be disabled in printer settings?

Maybe, i don't know. But as i said, when i run Flow Dynamics (which is basiclly same test) from BS it works

closed 2024-07-01 by github-actions[bot]