OrcaSlicer/OrcaSlicer #4686: UI - preview warnings appear behind gcode prevoew

trayracing, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

Version	10.0.22631 Build 22631

### Printer

creality k1 max

### How to reproduce

1. Ensure gcode preview is enabled in preview tab (like from a prior run)
2. In model view Import a model that produces warnings (see attached project)
3. Go to preview tab

### Actual results

* Warnings generated
* gcode preview obscures warnings, making them easy to miss and unreadable
Workaround: always disable gcode preview when first slicing, if you can remember how (shortcut "c", after searching web)
[image]

### Expected results

warnings should be top pane

### Project file & Debug log uploads

[OrcaBetaWarningsUnderGCODE.zip](https://github.com/SoftFever/OrcaSlicer/files/14736456/OrcaBetaWarningsUnderGCOD
[debug_Sun_Mar_24_14_17_03_26172.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14736462/debug_Sun_Mar_24_14_17_03_26172.log.zip)
E.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

closed 2024-06-30 by github-actions[bot]

magdalar, 2025-05-28:
For what it's worth, I found that "C" hides the G-Code window, allowing you to see the warnings behind. It's also accessible in the View menu.

mchant404-ship-it, 2026-09-16:
maybe if we start pressuring public figures in the 3d printing community to bring up how moronic this bug is, especially when you consider the extremely simple UI scaling fix they could apply to fix it, maybe they'll fix it? or should i just move over to prusaslicer like everyone else does in the end?

mchant404-ship-it, 2026-09-16:
> This is still a problem.
> 
> <img alt="Image" width="615" src="https://private-user-images.githubusercontent.com/231785/421441334-54e61f78-169a-4805-b7e9-0ec209c57df5.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3ODk1MzYxOTUsIm5iZiI6MTc4OTUzNTg5NSwicGF0aCI6Ii8yMzE3ODUvNDIxNDQxMzM0LTU0ZTYxZjc4LTE2OWEtNDgwNS1iN2U5LTBlYzIwOWM1N2RmNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwOTE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDkxNlQwNTE4MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YTkyOTIzMTE5ODA3NDFlZDRjNWRjYzg2YmJkMzA5YzE0Zjc0MzcyODQ1Y2NmNzM5NDUxZmNhMjgyMjM3MTU2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZyZXNwb25zZS1jb250ZW50LXR5cGU9aW1hZ2UlMkZwbmcifQ.wIsbCWnIJLxt4EKzCrnFfzHqfmzYHfyjapCa7d9u4is">
> Is there actually no interest in fixing this? I haven't be able to find a way to dismiss the Gcode to get at the warnings, which are not legibile. That seems like a problem!

seems like they dont realise this is such a crippling issue causing users to migrate over to new platforms. maybe need to get it into the public discussion?

trayracing, 2026-09-16:
> maybe if we start pressuring public figures in the 3d printing community to bring up how moronic this bug is, especially when you consider the extremely simple UI scaling fix they could apply to fix [...]

@mchant404-ship-it, if you know a fix, please create a pull request. Or, if you legally can't contribute code, point out where it went wrong and an example of where it was handled correctly. Being volunteers with limited time, they'd love to share the burden. 
I'm sure there's a more effective process to resolving this problem than a brusque comment on a closed issue.

mchant404-ship-it, 2026-09-16:
> [@mchant404-ship-it](https://github.com/mchant404-ship-it), if you know a fix, please create a pull request. Or, if you legally can't contribute code, point out where it went wrong and an example of [...]

Oh thats my bad i didnt realise it was volunteer run. No wonder it hasnt tried to get me to spend money yet.

I was completely wrong with my previous opinion, you folks are doing an amazing job. All things considered, im genuinly shocked its all so open source. This is exactly what 3d printing should be about, im just a dick. What language is it developed in? Or wheres the best place to learn the fundamentals of orcaslicers makings?

trayracing, 2026-09-16:
Assuming you're looking at this issue in github, the [< > Code](https://github.com/OrcaSlicer/OrcaSlicer) link near the top left corner of the page will take you to the top of the repo. That has a link to the source and will display the README for the project. The README explains a bit, like Orca's history and open source status. It also links to a wiki that includes build instructions and developer notes.