OrcaSlicer/OrcaSlicer #4690: Different filament in Orca: PLA appears as Support PLA (Sup. PLA) in 2.0.0-rc

eliezedeck, 2024-03-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-rc

### Operating System (OS)

macOS

### OS Version

Sonoma 14.2.1

### Additional system information

Apple M1 Max

### Printer

Bambu Lab P1P

### How to reproduce

1. Open Bambu Studio
2. Set filaments to any PLA
3. Open OrcaSlicer
4. Filaments will appear as Sup. PLA

### Actual results

As you can see, the filaments appear different in Orca Slicer since 2.0.0-rc
[image]

### Expected results

Obviously, the filaments should be the same as what appears in Bambu Studio

### Project file & Debug log uploads

This bug doesn't require an actual projet. But I did save an empty project for reference.
[Empty Orca project.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14738257/Empty.Orca.project.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

plexi32, 2024-03-26:
same on my X1C

sP33dAtTaCk, 2024-04-10:
Same problem on my X1C. PLA is displayed as “Sup.PLA”.
The problem has existed since version 2.0.0 (applies to Beta, RC and 2.1 Dev).
Since the wrong material is displayed, you cannot print directly.
The file must be saved in the cloud and then selected on the printer display :(

Zebble, 2024-04-25:
Can confirm the same behaviour since 2.0.0

plexi32, 2024-05-31:
still the same in 2.1 beta

CaptianNoob, 2024-06-06:
Problem still persists, but I was able to enable direct printing from the slicer by manually setting all the filaments to Bambu PLA instead of Generic PLA.

eliezedeck, 2024-07-14:
@SoftFever please, this is a major regression! I don't know if you are aware, but OrcaSlicer is just plain unusable on BambuLab printers since 2.0.0-rc and with the latest 2.1.1 (just tried now). I'm very sad I can't use OrcaSlicer unless I save the file on the printer first, which is a major productivity reduction.

ccinyadi, 2024-07-15:
@eliezedeck As @CaptainNoob just mentioned in the post above yours, instead of defining the filament as 'Generic' you can define the filament as Bambu PLA at which point, everything works through the interface as intended without having to save externally. Simple and easy workaround until the issue can be addressed.

valentinoConti, 2024-09-15:
Just downloaded OrcaSlicer 2.2.0-beta and same happens

I select "Generic PLA" in the Device > AMS section:
[image]

And then I get "ABS" (I never used ABS in my life):
[image]

Selection "Bambu PLA Basic" changes it to PLA correctly, BUT, the profile for Bambu PLA Basic it's not the same as Generic PLA; so for bad quality PLAs the Bambu PLA profile it's a mess... That's why Generic PLA exists, but for now its unusable

GrumpyGenius, 2024-09-27:
Using OrcaSlicer 2.2.0-beta:

I went into the profile of the Generic PLA and selected the tick box for 'support material'. This also appears to be a work around as by doing so it has allowed me to print from that AMS slot. The material name and display still show as 'Sup. PLA', but it is selectable for printing.

[image]

> @ccinyadi while the workaround is working, the fact that this is a regression is why it needs to be addressed.

Wholeheartedly agree, as others have stated, I too have had this issue since 2.0.0

SoftFever, 2024-09-28:
What happens if you set filament in Orca directly?

SoftFever, 2024-09-28:
Or vise-verse, set filament in Orca then check in BS?
Does it have the same issue?

manlaan, 2024-10-03:
Just a small addition that it is happening to me as well, with Generic PLA showing as ABS on my AMS  (same as valentinoConti).   When I edit the filament, it does show the correct value, as well as showing the correct value on the printer itself.  

As a work around, I've been setting it to one of the other PLA's.   (Overture settings are very close to Generic).

I checked in Bambu Studio and it works correctly there, but still not showing correctly in Orca.   I tried both ways, setting as Generic PLA in Orca (shows as ABS in Orca and PLA in BS), and setting as Generic in BS (shows as ABS in Orca and PLA in BS).  Also checked in Bambu Handy, which acts the same way as BS (correctly), and pulling up wrong in Orca.  

I also checked Orca beta2, and it acts the same way as beta1.  

It does this on all 4 spool positions.  

Orca Version: 2.2.0-beta
Printer:  X1C with AMS

RanaLoca, 2024-10-16:
I've been having the same issue since 2.0.0 released. Tried completely uninstalling orca slicer as well as everything linked to it and just installed 2.2.0-rc and I'm still having the same issue where generic PLA weather it be matte, silk, or normal, shows up in orca slicer as support PLA, causing anything that needs to be printed with generic PLA to error out.

Bambu studio displays the correct filament in the AMS, and trying to change the filament type in Orca slicer, bambu studio/handy, or directly on the printer does not change the outcome.

Running windows 11

aliorix, 2025-01-21:
Exactly same issue, Generic PLA showing like ABS in device tab, in the meantime chosing Generic PLA high speed is letting me print.
But is really annoying. Bambu studio, handy app showing the right filaments.
Win. 11

Ransom14, 2025-01-23:
Having the same issue as well.

PL4iNS, 2025-03-24:
Exact same issue with the new OrcaSlicer V2.3.0 build 70931e5....

Misharij, 2025-03-26:
Having the same issue as well

carpaij, 2025-03-27:
PolyTerra PLA appears as Polyterra Dual PLA after syncing the AMS in OrcaSlicer in V2.3.0 

[image]

[image]

SoftFever, 2025-03-30:
> PolyTerra PLA appears as Polyterra Dual PLA after syncing the AMS in OrcaSlicer in V2.3.0. Happens on both my P1S printers
> 

@carpaij Thanks, I can reproduce this issue. We'll fix it soon.

SoftFever, 2025-03-30:
Hey guys, I need your help to identify what other factors could cause such issues. I cannot reproduce it on my machine. Bellow is a video showing my attempt. My printer is in the LAN mode. I'm not sure if that will cause any difference.
Please let me know if you notice any difference in the reproducing steps.

https://github.com/user-attachments/assets/39b1e6f5-37e1-44b8-b117-56688b337b48

RanaLoca, 2025-03-30:
I'll add as much information as I can since I've been aching for this to be fixed for so long.

My printer is not on LAN mode, and initially I thought maybe it was because my X1C is running X1plus, but when I got my A1 mini, it is having the same issues.

I have Bambu studio installed as well of course, so idk if there are any shared files that might be causing issues maybe format wise.

I'm running on windows 11.

I honestly have no idea what might contribute to this because the issue persisted for me even after reinstalling orca, and even after building a different computer.

dartrax, 2025-03-30:
@Ice-Cue was able to overcome this issue by deleting the profiles/settings folder, reinstall orca and reimport from bambu cloud:

https://github.com/SoftFever/OrcaSlicer/issues/7432#issuecomment-2729182966

Unfortunately, we don't know what was wrong with the old config, I think it would be helpful if anyone could compare those folders before and after re-importing.

RanaLoca, 2025-03-30:
> [@Ice-Cue](https://github.com/Ice-Cue) was able to overcome this issue by deleting the profiles/settings folder, reinstall orca and reimport from bambu cloud:
> 
> [#7432 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/7432#issuecomment-2729182966)
> 
> Unfortunately, we don't know what was wrong with the old config, I think it would be helpful if anyone could compare those folders before and after re-importing.

I'll follow these steps, but I'll just rename the old config folder so I can compare it's contents once it is reinstalled. I'm not home ATM so it will be later today. I do hope this fixes it and we can figure this out for others

RanaLoca, 2025-03-31:
I followed the same steps as @Ice-Cue, but was unable to replicate their results. Generic PLA still ends up displaying Supp. PLA in the AMS and wont let me print with it. I have attached a video demonstrating the issue.

https://github.com/user-attachments/assets/fe4541a6-9b61-45d1-b37a-e9cb15550797

RanaLoca, 2025-04-03:
> I renamed the "C:\Users<username>\AppData\Roaming\OrcaSlicer" directory
> and reinstalled Orca.
> The problem seems to have gone away for me on doing this.

I don't know why that is not working for me... The only thing I can think
of is maybe something gets screwed up every time I import my filament and
machine presets from the cloud.

Smerks101, 2025-11-02:
> I renamed the "C:\Users<username>\AppData\Roaming\OrcaSlicer" directory and reinstalled Orca. The problem seems to have gone away for me on doing this.

Did you lose any settings doing this?

GCheung55, 2026-04-06:
I am seeing this as well. I have an X1C. I added an SV08, and the issue appears. Selecting Generic PLA displays Generic ABS and it's nozzle temps.

[image]

@SoftFever might this help reproduce the issue for your?