OrcaSlicer/OrcaSlicer #4691: Possibility to untie / disable inheriting between printer/filament/processing configurations

faith-healer, 2024-03-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

OrcaSlicer has some inconvenience with printer/filament/processing configurations - they're tied cascadically. If you want to move filament/processing configurations. For example, to your different printer (different root "ancestor") - you cannot, at least in easy way (yes, editing json files is not the most convenient way). And some settings are blocked being inherited - for example, when your selected printer is TwoTrees Sapphire SP-5 and you want to left only steel sheet texture - you cannot remove bed model (stl) except editing json file to remove that inheritance.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

1) make them at least as it done PrusaSlicer (backport).
2) make possible to untie settings from chain, maybe make a copy of configuration "ancestor".
3) something else in terms of realization but with suggested result.

Nazgul47, 2024-03-25:
Yes, current OrcaSlicer chained configuration system is very limited, changes suggested above are nice to be implemented.

Alexx40in, 2024-03-25:
I also want to note that when I change any printer profile.
For example I have selected Printer#1 and changed some processing settings (but does not save them).
I select Printer#2 and all current unsaved processing settings are gone by replacing by saved ones.

https://github.com/SoftFever/OrcaSlicer/assets/48121823/d6f199f0-1523-40ca-8fca-10818810cad6

faith-healer, 2024-06-28:
up?

closed 2024-10-05 by github-actions[bot]

faith-healer, 2024-10-05:
up