OrcaSlicer/OrcaSlicer #4694: Automatic bed mesh probe count problem.

LilleGill, 2024-03-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3 S1 with Sonic pad

### How to reproduce

Printing a longer slim item with probe count over and under 3 probe points.

### Actual results

Error on klipper screen about lagrange not supporting over 3 probe points. If I force start g-code to bicubic error states that lagrange should be used as one side is shorter that 4 probe points.

### Expected results

Probe be probing and print be starting.

### Project file & Debug log uploads

"code":"key47", "msg":"bed_mesh: cannot exceed a probe_count of 6 when using lagrange interpolation. Configured Probe Count: 7, 3", "values"

"code":"key239", "msg":"bed_mesh: bicubic interpolation with a probe_count of less than 4 points detected.  Forcing lagrange interpolation. Configured Probe Count: 7, 3", "values": [7, 3]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

pmogacs, 2024-03-26:
I have this issue too. lagrange also not good: bed_mesh: cannot exceed a probe_count of 6 when using lagrange interpolation. Configured Probe Count: 7, 3

vorrin, 2024-04-02:
Could it be this needs to be an && rather than ||? 

https://github.com/SoftFever/OrcaSlicer/blob/c663f4392587eea3f7e5153c72b6056c580f501f/src/libslic3r/GCode.cpp#L2342C1-L2344C10

cruiten, 2024-04-06:
> Could it be this needs to be an && rather than ||?
> 
> https://github.com/SoftFever/OrcaSlicer/blob/c663f4392587eea3f7e5153c72b6056c580f501f/src/libslic3r/GCode.cpp#L2342C1-L2344C10

The Klipper documentation indicates that Bicubic requires a minimum of four probe points along each axis, which means that the code is correct as written.

[image]

https://www.klipper3d.org/Bed_Mesh.html#mesh-interpolation

vorrin, 2024-04-06:
Oh I see, I was unaware of that, thanks! 
So it sounds like an 8,3 mesh is bound to fail with both lagrange and bicubic algorithms, wonder what the best approach to avoid this fail state is, whether to grow the mesh, increase the probe points, or just shoot a warning to the user at slice time

cruiten, 2024-04-06:
> Oh I see, I was unaware of that, thanks! So it sounds like an 8,3 mesh is bound to fail with both lagrange and bicubic algorithms, wonder what the best approach to avoid this fail state is, whether to grow the mesh, increase the probe points, or just shoot a warning to the user at slice time

Have you tried setting "mesh_pps" to "0" in your "bed_mesh" configuration setting?

vorrin, 2024-04-06:
no, but from the docs I also think that would technically solve it, but ... probably not the best possible solution? I'd have to remember this thing happens, make sure my mesh has a given amount of probes (which is quite opaque) and .. not have interpolation, so, I think at this point I'd rather fix the issue by somehow tricking the slicer into giving me a bigger mesh / rising the amount of probes manually

cruiten, 2024-04-06:
Have you tried having the slicer add a skirt?

That would increase the area to be probed/meshed...

vorrin, 2024-04-06:
agreed, that's one of the possible workarounds, though I suppose you can always imagine some geometry where this failure happens. So far I tended to just move objects around on the plate to fundamentally make a rectangle more than a thin strip. But for sure some sort of warning that this will happen, at slice time, would be handy, if nothing else (tho I think my ideal solution would be just growing the probe count to some safe minimums, we'll waste a tiny amount of time at mesh-generation but not have to worry about this, and have interpolation on our meshes)

LilleGill, 2024-04-06:
My workaround till now have been to print the object diagonally, or slight diagonally so the probe count will be 5+ in each direction.

cruiten, 2024-04-06:
Please be aware that in order to warn you the slicer would have to know several pieces of information:

1. It needs to know that you are using Klipper, which it knows from the "gcode_flavor" setting.
2. It needs to know that you have enabled adaptive meshing in your printer.cfg file for the "BED_MESH_CALIBRATE" command, which it currently does not have any awareness of...

It might even need more information from your "[bed_mesh]" definition in order to properly warn you...

vorrin, 2024-04-06:
@cruiten : re 2, I don't think that's true, adaptive meshing is enabled completely on Orca's own side and the information is sent for a correct mesh to be generated to klipper's vanilla 'BED_MESH_CALIBRATE' command. As [this ](https://github.com/SoftFever/OrcaSlicer/wiki/adaptive-bed-mesh) documentation confirms , there's no changes to be applied in printer.cfg, hence, I think the slicer knows the user has enabled adaptive bed meshing, as that is done explicitly in the printer settings within the slicer itself. 
Bit unsightly if the issue only happens on klipper and has to be fixed only for it, but I would imagine that's not the only time something like this happens.

cruiten, 2024-04-06:
@vorrin I see... My bed meshing is triggered within my PRINT_START macro, so that documentation does not really apply to me unless I updated my PRINT_START macro to consider an inbound parameter that determines whether I want the bed mesh to be adaptive.

To be clear, one needs to pass the "ADAPTIVE=1" parameter to Klipper's "BED_MESH_CALIBRATE" command in order to activate adaptive meshing.

But that is all good... The OrcaSlicer could determine whether adaptive meshing is active if it sees that any of the "Adaptive bed mesh" settings under "Printer settings" have been changed from their default values, or perhaps via a new checkbox in that section. Once it determines that adaptive meshing as desired then it could possibly provide the warning that you requested.

It sounds like this is turning into a new feature request... ;-)

vorrin, 2024-04-07:
eh, imo not quite, I mean... pretty sure it was never anyone's intention of random prints getting your printer stuck (in my case, that also happens in a totally non-polite way, printer just freezes where it is, all temps full blast, and stays there cathatonic but power hungry), I'd say this _feels_ a lot more like a bug/unexpected behaviour needing fixing, than a feature :)

cruiten, 2024-04-07:
All I was saying is that not everyone uses adaptive meshing, so warning people for an adaptive meshing related issue would have to be conditional...

pmogacs, 2024-04-08:
Simply expanding mesh area looks cleaner no warnings, moving, skirting and other workarounds needed. Also for someone not using adaptive mesh, nothing to worry about. I think from UX perspective this is the best. Maybe from development point of view is also easier, just throw another probe point in if it's not enough.

vorrin, 2024-04-08:
Yes, agreed, I would personally solve it like that, as the slicer has all necessary information, and yeah that's the smoothest UX

igiannakas, 2024-05-07:
A PR with a potential fix has been posted - please test the code build there and report back

igiannakas, 2024-05-07:
> Oh I see, I was unaware of that, thanks! So it sounds like an 8,3 mesh is bound to fail with both lagrange and bicubic algorithms, wonder what the best approach to avoid this fail state is, whether to grow the mesh, increase the probe points, or just shoot a warning to the user at slice time

The fix implemented is to increase the probe points to 8x4 in your above example. This is the approach that KAMP is also using to derive the bed mesh points. Its better to get a higher density mesh than expand the mesh boundaries.

closed 2024-08-13 by github-actions[bot]