OrcaSlicer/OrcaSlicer #4695: Custom notification - at layer pause (Bambu Lab)

tlhintoq, 2024-03-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not related to a bug

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Bambu Lab machines like the X1 Carbon or P1P will raise a notification from the printer to the app on your phone.  So you get an alert like "Job completed Successfully" or "AMS failed to retract material" etc.

It would be great if we could get a message at layer pause, and thus the app on the user's phone would ping them that the machine had paused and they should come do whatever is they need to do.

[image]

kylek29, 2024-03-25:
This would require firmware changes by BambuLab to implement a custom GCode that triggers their notification function.

tlhintoq, 2024-05-10:
Bot nudge.

closed 2024-08-17 by github-actions[bot]