OrcaSlicer/OrcaSlicer #4696: Add "Export as... BambuStudio" would be nice so Orca users could contribute to MakerWorld.

tlhintoq, 2024-03-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If a job is for a BambuLab printer it would be nice to export the 3mf such that it is recognized as a BambuStudio file so users can contribute to MakerWorld.

[image]

[image]

DannyH3103, 2024-04-04:
I would love to see this feature too.

tlhintoq, 2024-04-15:
Keep alive.  Problem still exists in current 2.0 nightly as of 15apr2024

tlhintoq, 2024-05-10:
Bot nudge.

Kaympe20, 2024-06-29:
Seconding this.

closed 2024-10-05 by github-actions[bot]

Kaympe20, 2024-10-05:
Bot nudge.