OrcaSlicer/OrcaSlicer #4709: Floating lines on Overhang Walls with Inner/Outer/Inner sequence!

Vitorleoron, 2024-03-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-rc

### Operating System (OS)

Windows

### OS Version

11

### Printer

K1 Max

### How to reproduce

When I printed some object that have some overhang walls, with this settings above:

[image]

When it reaches the layer that start close the hole, it prints the Inner wall of the bridge/overhang wall first then the "Outer" and the Inner, so it prints this lines floating, as the video above:

https://github.com/SoftFever/OrcaSlicer/assets/15314940/f5762a62-bfd3-4faf-a8e2-df5c3780132e

### Actual results

The results in the print is this:

[image]

### Expected results

This bridges should be printed in the Inner/Outer order to be anchored by the other lines!

### Project file & Debug log uploads

[debug_network_Mon_Mar_25_17_44_52.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14750192/debug_network_Mon_Mar_25_17_44_52.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-03-27:
Does it happen in 1.9.1 or 2.0.0-beta too?

Vitorleoron, 2024-03-27:
Yes, worst in the 2.0.0-rc version, but in the overhang wall lines the problem still occur in the old versions too, in the 2.0.0-rc the floating lines happen more often

SoftFever, 2024-03-28:
> Yes, worst in the 2.0.0-rc version, but in the overhang wall lines the problem still occur in the old versions too, in the 2.0.0-rc the floating lines happen more often

Could you share the project? I'd like to see if there's been any regression. 
Meanwhile, based on the photo, this model has very steep overhangs. 
It's not recommended to use a sandwich or outer wall-first sequence for such models.

Dustmuffins, 2024-04-28:
> Could you share the project? I'd like to see if there's been any regression. Meanwhile, based on the photo, this model has very steep overhangs. It's not recommended to use a sandwich or outer wall-first sequence for such models.

I'm not the OP, but I was able to recreate it on this file.

I'm not sure if this is the best place to request this, but is it possible to request a feature to switch to inner/outer w/ precise walls if the overhang is this extreme?

[large overhang.zip](https://github.com/SoftFever/OrcaSlicer/files/15141295/large.overhang.zip)

tastyratz, 2024-07-06:
Does this happen only with Arachne or with classic walls as well?

pavlokomarov, 2024-07-29:
@SoftFever is any updates? Faced with similar issue: outer wall overhang is printed first, despite Inner/Outer order. Orca slicer 2.1.1

https://github.com/user-attachments/assets/848909a3-0325-44d0-868d-e7b837000f2d

FelixBuehler, 2024-09-30:
Can confirm this issue. Setting is Inner/Outer, but the overhang wall gets printed first.
[image]

closed 2025-01-06 by github-actions[bot]

Maximus-CZ, 2025-03-08:
Yo its stale and closed but the issue still persists.. Ideally I want to keep Inner/outer/inner, but overhanging parts should still be printed with inner/outer to prevent printing midair..