OrcaSlicer/OrcaSlicer #4710: 2.0.0 Scarf, infill pattern, and some other settings never saved

xTremePower, 2024-03-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 beta

### Operating System (OS)

Windows

### OS Version

11

### Printer

No matter

### How to reproduce

Change Scarf joint settings, save the profile, close Orca, launch it again, settings reset
It do the same with many other settings, like infill pattern,  scraf joint settings, line width ...

### Actual results

This setings always reset to default after saving the profile and closing Orca ... 
No matter if you save them into a new profile name or old one 

### Expected results

save the profile to not  need to modifiy all those settings after All restard of Orca

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14750226/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

vgdh, 2024-03-26:
have you tried a new one release candidate version?
https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds

xTremePower, 2024-03-26:
Yes I tried, it's the same problem

SoftFever, 2024-03-27:
I can't reproduce it either.

Can you provide a screen recording to show case what did you do?

xTremePower, 2024-03-27:
Yes, I will show you the process today or tomorrow in a screen record.

xTremePower, 2024-04-03:
I just tried in 2.1.0 dev .. still make me same .. it drives me crazy ...

Even worse : when I hit Export - Config, it directly crash Orca compeltely ....

I will record thse bugs and uplaod on youtube to show you

xTremePower, 2024-04-03:
Here are the record
I made it in 2.1.0 dev but it's exactly same in 2.0.0 /rc/beta/normal

So now I can't even export all my settings to save them and make a completly new and clean install and reimport all.
And as you see I have many profiles, I really don't want to loose them all ...
Thanks for your help 

https://youtu.be/E8sLyx-5Ts8

What files can I provide you to help you

Ps : as I said, it never made me this in 1.x.x versions at all

xTremePower, 2024-04-03:
I don't know why, I don't even see these settings in the json file..
Here is a freshly saved profile with 0.8mm outter wall, scarf enabled, lightning infill,precise wall ... all these settings for example are never saved now ...
Before 2.x.x for exemple, in this profile, the precise wall was always enabled and there after orca reboot, also the lightning infill, who now rester to cubid after orca reboot
[Mingda 0.6mm TPU Hard.json](https://github.com/SoftFever/OrcaSlicer/files/14858669/Mingda.0.6mm.TPU.Hard.json)

xTremePower, 2024-04-05:
I just did an experiment:
-I have orca 2.0.0 installed with all my profiles
-If I try to export the profiles in the application, Orca crashes directly.
-If I save settings in a profile and restart Orca, it forgets SOME settings, but not others...
-I renamed the OrcaSlicer folder to OrcaSlicerBAK in roaming,
-I restart Orca WITHOUT REINSTALLING IT, I configure a bogus printer, I save the settings in the profile (especially those which were previously forgotten)
-I restart Orca, and the settings are still there! I click on Export Config, and it works perfectly!
-I leave Orca, I delete the new OrcaSlicer folder in roaming, I put the old one back
-I restart Orca without reinstalling, and the export problems and forgetting settings come back directly....

So the problem does not come from the installation of Orca, but from something, or I don't know what file to put in the OrcaSlicer folder in roaming...

I think I'll try putting the json files back one by one in a new OrcaSlicer folder in roaming, and restart orca each time, and see after which file everything will start messing up again. I only see this as a solution to find where it comes from

xTremePower, 2024-04-05:
If you want @SoftFever I can send you in private (or with a google drive link (in private) my problematic OrcaSlicer folder from roaming

SoftFever, 2024-04-07:
@xTremePower 
Can you try with other printer profiles?
It looks like you are using a custom profile, it's possible that the profile has issues

xTremePower, 2024-04-07:
Hello,

I tried multiple profiles, but I stopped play with because when I open a
profile he forgets some settings.. So slowly I destroy other profiles
too....

And it not explain why with this OrcaSlicer config folder it make 100% of
time orca crash when I hit the export config menu 😢

SoftFever, 2024-04-07:
I'm sorry to hear that. I haven't been able to replicate the issue, and there haven't been any other user reports. Diagnosing it remains a challenge.

closed 2024-07-14 by github-actions[bot]