OrcaSlicer/OrcaSlicer #4716: Cooling - minimal layer time - wait instead of printing slowly

DiegoTheWolf, 2024-03-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am printing parts that have a rib on top of them. To get the layer to cool down enough (slow_down_layer_time) the print will be slowed down to slow_down_min_speed. 
Problem: My material looks different when i reduce my print speed to much.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If it could keep the print speed up, retract and wait at the outside of the part (away from the structure to not melt anything).
If there is a small ooze this could be cut away, but my part would look the same up to the top.
Adding a boolean parameter to control this would give flexibility to everyone.
If activated the slow_down_print_speed would still be active, but the time the layer needs to wait to reach slow_down_layer_time would be waited outside of the part.

### Describe alternatives you've considered

Currently there is no alternative.
Print super slowly or decrease layer time.

vgdh, 2024-03-26:
but ozzing will ruin the print. After waiting printer needs some little purge and prime.

KathrynHazuka, 2024-03-28:
Im not sure what the solution to this is, but the slowing down of the print messes with my layers and they look so drastically different that it ruins the look of the part. It's impossible to have a clean finish with this setting on, but without it, the part looks bad in other ways.

closed 2024-07-05 by github-actions[bot]

zoltan0013, 2025-11-11:
I remember i use this function in Creality Slicer and maybe in OrcaSlicer too. I realy miss this function.