OrcaSlicer/OrcaSlicer #4717: Add free floating parts

DiegoTheWolf, 2024-03-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have free form part that has to be fitted somewhere.
When i want to print it normally one point always has to touch the buildplate.
The part - due to is curvature - is build fully on supports.
However since (in normal orientation) all 4 corners touch the ground it's sometimes tricky to remove.
Removing the support is easy though.
If it would had support all around, the part would be better to print.
Currently i have to create a "sacrificial cube" (add part) after importing my part. this cube sits at zero height and will be removed (add negative part). This construct then allowes me to raise the original part 1mm to have the support structures around.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Be able to place parts not only on faces but give them a additional z-height so they are free floating (like SLA printers).
Add a parameter for "additional z-height" so we don't mess with the basic code for positioning.
This way you could use the lay down tool to have some face oriented at 0°, then orient it the way you need and lift the print to be printed fully on supports.

vgdh, 2024-03-26:
may be using raft is the solution?

levleon13, 2024-03-27:
i'd like that feature too

DiegoTheWolf, 2024-03-27:
> may be using raft is the solution?

No, it's just easier to remove a millimeter or two of supports.
If the supports are at the edges you only have one layer of support and this can really bind with the part.

closed 2024-07-03 by github-actions[bot]