OrcaSlicer/OrcaSlicer #4723: Un-called sudden flow and speed changes where speed and flow should be uniform

Rickthebig, 2024-03-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

DIY CoreXY 3D printer (BLV Cube) based on Troodon (RRF)

### How to reproduce

Load the attached project (an egg form printed in spiral vase mode) and look at the bottom layers as in the pictures.

### Actual results

Both flow and speed change unexpectedly while building the first layers. The change is very obvious while looking at the piece during printing time. These are planned changes in the slicers, so the defective lines can be recognized:
[image]

[image]

[image]

[image]

[image]

There are no other conditions like overhangs, gaps, geometric irregularities etc. that would explain these changes. A sudden flow change is generally undesireable.

### Expected results

The layer lines should be laid with regular, constant speed.

### Project file & Debug log uploads

[Easter_Egg_PATTERN_-_KRAKEN.zip](https://github.com/SoftFever/OrcaSlicer/files/14758341/Easter_Egg_PATTERN_-_KRAKEN.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

See also related issue:
Inconstant flow #1360

vgdh, 2024-03-26:
may be related https://github.com/SoftFever/OrcaSlicer/issues/4634

vgdh, 2024-03-26:
I've disabled **small area flow compensation** and your project sliced with consistent flow.

vgdh, 2024-03-26:
monotone line helps as well

Rickthebig, 2024-03-26:
> I've disabled **small area flow compensation** and your project sliced with consistent flow.

OK, then let's ditch this "small are flow compensation" feature :-)

igiannakas, 2024-03-29:
I’m guessing you have extrusion rate smoothing enabled? If so this is to be expected as together with small area infill compensation you get variations in flow that need to be compensated for.

perpleksed, 2024-06-12:
> I’m guessing you have extrusion rate smoothing enabled? If so this is to be expected as together with small area infill compensation you get variations in flow that need to be compensated for.

Resulting flow is not desired though, I think, maybe Small Area Flow Compensation should be applied **after** Extrusion Rate Smoothing? Look at this example:

ERS only (looks reasonable):
[image]

SAFC only (flow adjustments not even visible here):
[image]

ERS+SAFC (wow, suddenly we are barely moving at 60mm\s for solid infill!):
[image]

igiannakas, 2024-06-12:
I think you’ve got your ers value a bit too low. In any case small area flow compensation together with ers works best with monotonic lines, not monotonic due to the constant small segments that monotonic generates.

closed 2024-09-18 by github-actions[bot]