OrcaSlicer/OrcaSlicer #4726: overhang speed settings ignored

jamincollins, 2024-03-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-rc

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Additional system information

N/A

### Printer

RatRig 400mm

### How to reproduce

1. Grab: https://www.printables.com/model/126029-overhang-test
2. Slice it
3. Check speeds

### Actual results

[image]

### Expected results

That the selected speeds for a given overhang percentage are actually respected.

### Project file & Debug log uploads

[logs.zip](https://github.com/SoftFever/OrcaSlicer/files/14763555/logs.zip)
[overhangs.zip](https://github.com/SoftFever/OrcaSlicer/files/14763558/overhangs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-03-27:
It's is percentage.
Can you clarify which part do you think is not right?

Please note that it use unsupported area of a perimeter, not angles.

jamincollins, 2024-03-27:
If you look at the screenshot you'll see that the area marked as a 30 degree angle has a speed indication of roughly 158, but the settings for that angle indicate it should be 60.  Likewise for all the angles.

jamincollins, 2024-03-27:
> It's is percentage. 

The dialog indicates **mm/s _or_ %**.  I haven't put a **%** after any number.  So, I would expect it to be **mm/s**.

lhndo, 2024-03-27:
Looks like you also have unwanted slowdowns in non overhang places that will cause artifacts.
Like reported here: https://github.com/SoftFever/OrcaSlicer/issues/4553

SoftFever, 2024-03-28:
> If you look at the screenshot you'll see that the area marked as a 30 degree angle has a speed indication of roughly 158, but the settings for that angle indicate it should be 60. Likewise for all the angles.

Like I explained, the algo don't use surface angle to calculate the overhang. 
The speed is linearly interpolated between 60mm/s  and 200mm/s, as you set 0 for "10% 25%" which mean no slow down.

SoftFever, 2024-03-28:
Re-open it as it was accidently closed.

igiannakas, 2024-03-29:
The %age you see in the brackets in the overhang speed section is not an angle but rather how much overlap the extrusion lines have between each layer. Inspect your sliced model and you should see the overlap corresponding to the selected speeds.

jamincollins, 2024-03-30:
I'm having a great deal of trouble printing the above model beyond the 40 mark. Overhangs are terrible and ultimately result in the nozzle hitting the print and detaching.

If these aren't the settings to allow the print down in these cases, what settings should I be looking at?

SPYKE357, 2024-05-06:
i agree with the OP. the speeds do not match the overhang percentages, the first 3 percentage ranges give you more control over low angles ( yes i know the percentages arent angles, but they can be calculated using arctan) whereas 75%;100% literally will dictate the speed from 48 degrees to 90 degree overhangs. the logic on overhangs is completely backwards. more speed control is needed on the higher overhang angle ranges (from 50% to 100%)exactly due to arctan calculation. these should be far more incremental when applied correctly.
Check evidence attached. 5mms is implemented from the 50 degree mark upwards, when it should only be implemented from 64 degrees onward. meaning that 75%;100% range should be controlling about 26 degrees of overhang, but currently controls about 42 degrees as per model.
model is 0.16 lh and 0.42 lh. 
Overhang % does not give control where the control is needed most. the 75
[image]

according to arctan  rough calcs at 0.16 and 0.42
0%-25% should give us control of 0-34 degrees
25-50% should be 34-53 degrees
50-75% should be 53-64 degrees
75-100% should be 64-90 degrees.( but currently affecting 48 to 90 degrees)
[image]

clearly something is not right with overhang percentage implementation and how the arctan results correlate with the respective percentages.

as per picture. the 5mms is being implemented at 48 degrees and has been verified by double cheking fopr g1 f300 commands at the overhangs, according to my settings full speed print should be implemented right up to 64 degree overhangs.

you can all double check this on your orca builds and slicing a overhang test model. set all ranges to 0 except 7%;100% at whatever speed will show up clearly for you on slicer preview, i used 5mms.

SPYKE357, 2024-05-06:
sorry to necro the post but attention to this is still needed

SPYKE357, 2024-05-06:
I've been investigating this issue and also thought there may be an issue with overhang speed implementation but i had slow down for curled perimeters setting on as well as the overhang speed control settings active.

having done some sanity checks with arctan calculations and double cheked settings, the overhang speed implementation by SoftFever is spot on according to the ranges.
[image]
according to the arctan calcs the speed/angle implementation is spot on.
 however please can we have smaller 5-10% increments between 75%-100% overlap range. @SoftFever im sure you've noticed the range of degrees controlled by each of the overhang % ranges. Using overhang % as the main factor to control speed gives you more middrange control over speeds but not where they are needed most which is the 75%-100% overlap range, eg 0.16 lh with 0.2 lw with 75% to 100% overlap is a range of +-26 degrees, 50% to 75% is a range of +- 11 degrees, 25% to 50% is a range of +- 19 degrees and 0% to 25% is a range of 34 degrees{ this range of degrees in this parameter is perfect/good enough for the angle range} though you can see that the range for 75% to 100% is too broad/blunt where it matters the most. 

May i also suggest an editable speed parameter for slow down for curled perimeters speed? or scaling factor according to angle/overhang %?
would be great.

jamincollins, 2024-05-06:
> although a grey pop up showing the angles controlled in the overlap %'s would be more user friendly for some, saving time to whip out the calculator or slice and preview for angles affected?

SPYKE357, 2024-05-06:
Self explanatory

closed 2024-05-18 by jamincollins