OrcaSlicer/OrcaSlicer #4732: Travel over part is leaving SCAR

inspectionsbybob, 2024-03-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am always frustrated when It is almost finished a beautiful top surface...
And it travels over it, leaving a scar on an otherwise perfect top.

Is there a way to do a Zhop on TRAVEL so when it goes over something it is not touching?

Is "Do not cross walls" counter to this as it can't escape to go around?

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a new option of "Lift on Travel"  that would not be a Zhop which can exacerbate Strining.  It would be separate, only for Travel.

Another option would be to "go around" even if "don't cross walls" is turned on....

levleon13, 2024-03-27:
zhop fixed it for me. When your nozzle travels it retracts, so I think you're asking for something that exists. 
lift on travel = zhop

inspectionsbybob, 2024-03-27:
Zhop is great for travel, but often increases stringing, or if you use one
of the advanced (spiral) retraction settings it can slow things down (a
lot) on retraction heavy models.

I was/am suggesting a "different" retraction for travel.  I have stringing
pretty much dialed out, but at the cost of top scars.

The other option is something like "do not cross walls" that would force it
to "go around" rather than through... maybe it could have a minimum travel
distance and some other tests to make sure the scar would be large enough
to be worth the extra travel.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

levleon13, 2024-03-27:
> Zhop is great for travel, but often increases stringing, or if you use one of the advanced (spiral) retraction settings it can slow things down (a lot) on retraction heavy models. I was/am [...]

Tbh I am running between 200-300mms so I guess I miss that stringing issue. My normal zhop
Mode has 0 strings

Rickthebig, 2024-03-28:
> The other option is something like "do not cross walls" that would force it to "go around" rather than through... maybe it could have a minimum travel distance and some other tests to make sure the scar would be large enough to be worth the extra travel.

Increase the "Max detour length" then. It will avoid crossing borders when possible.

Rickthebig, 2024-03-28:
> or if you use one of the advanced (spiral) retraction settings it can slow things

I could never get spiral retractions to work (2.0 Beta). Neither slope retractions. The retraction fraction before wiping is always 85% if set 85% or below. 

How did you get spiral retractions to work at all? What settings do you have regarding wipe, z hop, and retraction?

inspectionsbybob, 2024-03-28:
Spiral worked, but it made a big spiral, and didn't seem worth the extra
time.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

igiannakas, 2024-03-29:
You can enable zhop on top surfaces only. This should address the top surface scarring you’re seeing.

inspectionsbybob, 2024-03-30:
igiannakas
Where is an option for Z-Hop on Top Surfaces only?  I looked in Printer, Filament and Process and could only find:
Auto
Normal
Slope
Spiral

closed 2024-05-24 by SoftFever