OrcaSlicer/OrcaSlicer #4735: Slope speed control above infill

levleon13, 2024-03-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When printing at 250mms, 0.3 layer height, internal slopes are hard to fuse with the model, resulting in bad adhesion and bad finish.

### Which printers will be beneficial to this feature?

Klipper, Marlin, Others, All

### Describe the solution you'd like

This is a very much desired function for faster speeds. When my slopes go inward, at 250mm the nozzle can't lay a good internal solid infill to support the slope walls ( at 0.3mm petg_cf). Fixing it with much denser infill wastes time, money, and logic. 
it would be amazing to control the speeds of inside sloped walls, solid infill, or even better,  both. 

### Describe alternatives you've considered

my workaround is to reduce the layer height and overall speeds, but this is not the optimal way to achieve this.

igiannakas, 2024-03-29:
Could you post a picture of the sliced file in the problematic regions? It’s hard to understand the issue without a model file or a picture.

levleon13, 2024-03-29:
> Could you post a picture of the sliced file in the problematic regions? It’s hard to understand the issue without a model file or a picture.

[image]

this is an inward slope that has no speed controls. ( this is not very extreme slope but just for reference )

igiannakas, 2024-03-29:
Have you tried setting the don’t filter out small internal bridges to no filtering and ensure vertical shell thickness to all?

closed 2024-07-05 by github-actions[bot]