OrcaSlicer/OrcaSlicer #4736: [FR] Orthographic View

foreachthing, 2024-03-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

While arranging tall objects manually, it was always helpful (in PrusaSlicer) to enable orthographic camera instead of perspective.
In Prusa, there is a convenient keyboard shortcut as well.

### Describe alternatives you've considered

Well, short from arrange it in Prusa and then open it in Orca, there is really none.

### Additional context

Thanks for considering.

SoftFever, 2024-03-27:
While there no hotkey to switch between perspective and ortho, we do support ortho view.
You can switch through the menu:
[image]

foreachthing, 2024-03-27:
I did not see that 🙄 🙈 
Thanks!!