OrcaSlicer/OrcaSlicer #4737: Weird intersection between top layer and following wall

martinrosolino, 2024-03-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Ender 2 Pro

### How to reproduce

Load a model with any kind of step, preferably in a rounded shape, set Ensure Vertical Shell Thickness to Critical Only (for every other print feature it works amazing, but in this particular case I've to set it to All to get a nice result, wasting time/material on every other place it's being applied)

### Actual results

[image]
the top layer "ends" before the wall, so it leaves a tiny gap

### Expected results

Maybe add a parameter "Top layer - Next wall Overlap"?

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gregmatic, 2024-05-22:
i came here to ask about my own issue but spotted your post first. i wonder if we are having the same issue? you'll have to look at the slicer preview to tell. orca is not slicing a full top layer if the next layer has structures above it. 
[image]

[image]

martinrosolino, 2024-05-22:
I don't think that's exactly the same issue, but yeah results should be similar. In your case, the top structure is rather thin so a full "red" layer would be a great option, in my case I don't think it makes sense to print a full layer before a big top structure, I believe that about 2mm in is enough, that's what I get when I set Ensure Vertical She'll Thickness to All, but that doesn't give me good results in other parts of the print.

MaroonOut09, 2024-05-28:
I'm having the same issue, and, in my case at least, it looks like the top infill isn't extending to the interior perimeters.

[image]

When I slice the same stl with the same settings in PS or SS, this gap isn't present.

foreachthing, 2024-05-29:
Could you please upload your models? I cannot reproduce with cubes and cylinders.

foreachthing, 2024-05-29:
I don't see any of that issue. But, again, you should upload your 3mf file (zip it first).

[image]

[Mold_cactus_n10_nucleo.zip](https://github.com/SoftFever/OrcaSlicer/files/15489522/Mold_cactus_n10_nucleo.zip)

MaroonOut09, 2024-05-30:
[Meter Base-RC2.zip](https://github.com/SoftFever/OrcaSlicer/files/15492666/Meter.Base-RC2.zip)
Attached is the stl I'm experiencing the issue with.

foreachthing, 2024-05-30:
Please attach your *.3mf (zip it first). That way, we can all see your settings, the orientation of your model and everything else relevant.

MaroonOut09, 2024-06-02:
Attached is the .3mf as requested.
[Meter Base-RC2 (2).zip](https://github.com/user-attachments/files/15523774/Meter.Base-RC2.2.zip)

closed 2024-09-09 by github-actions[bot]