OrcaSlicer/OrcaSlicer #4745: Dynamic retraction length

Rickthebig, 2024-03-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

A fixed Retraction length is a somewhat inadequate solution of the well known problem of oozing. If the pressure in the melt chamber of the nozzle is high just prior to retracting, the length should be rather long, because the filament would ooze more. But is the pressure always high when doing retractions? 

It is not.

That's why a fixed retraction is not suitable for all situations. Either over-retraction will take place where the nozzle is moving slowly and deccelerating, or it will be not enough after a high acceleration, after fast moves more so. In any case, oozing might occur if the travel following the retraction is long.

Suppose that the model has a lattice surface, like made of wires. The nozzle would move from one point, where it deposits just a bit of filament, to the next, which is very close. This case is tipical for a small retraction length. Currently, since no dynamic retraction is implemented, the extruder stepper will unnecessarily move the filament back and forth to the configured (too long) retraction length, slowing the print and perhaps grinding the filament. These unneeded long retraction/deretraction movements in swift succession also introduce additional variation of the extruded  amount, causing extrusion inconsistencies.

This feature request proposes 

1. a way to avoid over-retraction
2. a method to guarantee that retraction length is adequate even in borderline cases

by adapting the retraction length dynamically during the print process.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

To implement the dynamic retraction, the slicer should use a retraction factor (provided by the user), which does not represent a "length", but the general propensity to ooze. Then the slicer analyses the course of filament deposition. Taking speed and acceleration of the previous moves since last retraction (which are accessible to the slicer software), and taking into account the length of travel following the retraction, a simulated pressure index is calculated, which multiplied by the given retraction factor would determine the actual retraction length. 

Higher accelerations and/or higher speeds since last retraction prior to the upcoming retraction would consequently command a longer retraction length. Also longer travel movements following a retraction would also need a rather longer retraction length. Both conditions (simulated pressure and travel time) should be considered when calculating the dynamic retraction length.

Slow flow, specially when deccelerating, would imply a shorter retraction length. If the next de-retraction is near, there is no need for a long retraction. Similarly, a subsequent short travel movement needs only a short retraction to avoid oozing.

The frequent problem, that the retraction length determined with a calibration test is too short and does not prevent stringing during actual printing, would not occur anymore. Retraction calibration models are mostly small so the result is relevant to small models and does not account for long travel movements in large prints. A dynamic retraction would solve that problem.

### Additional context

Implementing this feature would redefine classical retraction tests: Not the retraction length is the variable being changed layerwise, but an artificial retraction factor. Any test aimed to determine this factor should be designed including different speeds, accelerations and travel lengths. New calibration models would be then needed.

vgdh, 2024-03-28:
May be even change retraction speed:
1. **Longer travel** - short fast retraction at place flowed by long slow retraction and simultaneous movement to next print point at the same time.
2. **Longer travel** - only  short fast retraction

Rickthebig, 2024-03-28:
> May be even change retraction speed:
> 
> 1. **Long travel** - short fast retraction at place flowed by long slow retraction and simultaneous movement to next print point at the same time.
> 2. **Short travel** - only  short fast retraction

I mainly use Cura (although I love Orca, but it is still too buggy) and there I use a postprocessing script will all models I print. It is called "Retract Continue" which adds an additional travel-dependent retraction length that is executed during the travel. It accepts only one parameter: The percentage of travel (resulting in the additional retraction). At the target, the same amount is then unretracted.

Something similar could be also implemented in this dynamic retraction idea.

igiannakas, 2024-03-29:
Does your printer have pressure advance?

If it does this feature does exactly what you’re looking for. It retracts the filament reducing the pressure in the nozzle as the hot end comes to a stop. Then retraction is needed for only a very small amount to prevent that final bit of oozing and stringing (I use 0.3mm retraction on my PA tuned machines with no stringing).

this feature is best implemented in the printer firmware as it knows exactly the speed of the hot end. The slicer can only approximate it.

Rickthebig, 2024-03-29:
> Does your printer have pressure advance?

Yes I have tuned pressure advance (RRF). As a consequence, the retraction length of my  Bowden setup (60 cm tube length) was reduced from 6 to 1.5 mm. Pressure advance is a bless.

But retraction still has to be set, and I mean that.

This is the first time I read that pressure advance replaces retraction. It does not. Your understanding of pressure advance is rather questionable. 

As the filament is extruded, pressure advance dampens pressure changes inside the melt chamber a bit (not completely, otherwise acceleration/decceleration would not work well). But most important, it anticipates such changes, compensating for the overall compressibility that leads to downstream over- or underextrusion if left disregarded. 

But there is still a **static** pressure. The pressure is not reduced to zero at the end of a loop and it does not start from zero at the beginning of the next loop.  The higher the average extrusion flow, the higher the static pressure. Static pressure is high between two fast loops and low between two slow sloops. Exactly at these points, retraction kicks in, classically without distinguishing those two different cases.

Retraction is still needed to overcome the effects of this static pressure at travel time.

igiannakas, 2024-03-29:
Where did I say it replaces retraction??? I said it reduces the retraction distance needed as it reduces the nozzle pressure… 

You can set your retraction distance to be the one needed to support long travel moves. Then you don’t get oozing on long travels. Then, your smaller travel moves still function well as while they don’t need as much retraction to combat oozing from travel, a longer retraction distance doesn’t hurt either 

The only downside is a slight increase in print time. Finally you can’t keep increasing retraction to combat oozing because after a point the nozzle will suck air in causing pockmarks in the print. See below for the artefact that can result from too long retraction that causes the nozzle to suck air in:

[image]

In addition, the material flowing down the nozzle during long travel moves due to its viscosity, will result in a slight extrusion inconsistencies when starting the next perimeter as the nozzle will start flowing slightly before the complete de retraction distance is achieved. This shows at the seams when printing external perimeter first as slight over extrusion. See below:

[image]

The way to mitigate this is by starting the extrusion after a long travel move in an internal perimeter so it’s hidden. This is what the wipe before external perimeter feature that I’ve implemented does and helps with. Also why printing with inner outer inner mode gives better seams compared to outer inner.

So the way to address this in its root, is to travel as fast as possible with the highest acceleration & jerk possible giving the nozzle as little time as possible to flow uncommanded. In addition tune your retraction to the highest distance and speed before you get pockmarks like the above picture. You can tune these per filament (so petg would have a higher retraction than PLA etc). 

Finally tune your wipe before retraction distance in the printer settings to ensure that any back pressure is alleviated before travelling. This retraction while wiping is done at the last print speed so it serves as a mechanism to further reduce pressure build up depending on the print speed you were going before retraction.

That way your nozzle pressure is always 0 irrespective of travel distance, previous print speed and extruded flow rates. 

That way you get better seams as there is less uncontrolled flow before the deretraction move, a consistently 0 pressure remaining in the nozzle and less ooze within the limits of your machine.

Rickthebig, 2024-03-29:
> You can set your retraction distance to be the one needed to support long travel moves. Then you don’t get oozing on long travels. Then, your smaller travel moves still function well as while they [...]

Try to print a Voronoi model and you will understand why unnecessary long retractions are not good. Many "torture" models are designed to challenge the ability to extrude discontinuously.

But most important, you might be underestimating the evil combination of pressure advance with too long retractions. I have witnessed weird filament movements and hammering extruder steppers under those circumstances, even a continuous retraction where the filament is just rolled back all the way out the PTFE tube (it is a kind of accumulating effect). Out of my astoundingly uncertaintyI have done some research and found similar cases. That is why in all flavors of 3d printing firmware you are insistently advised to revisit your retraction settings when using pressure advance. They both tend to interfere. But still, retractions are needed, except if Orca developers are pleased with a mediocre performance.

And I recommend to take a look at the "Retract Continue" postprocessing script available for Cura. It allows you to split the retraction length into an immediate fraction followed by a continuous retraction along the travel path proportinal to its length (up to a limit). I have had very good results with it.

Thanks for your hints. I will engage with wipe tuning when it actually works as it should... Currently, the "retract before wipe" fraction is always 85%,  when this fraction is set to less than 85 (2.0 Beta). There are too many combinations of wipe, z hop types (also flawed), retraction length and speeds, and where travel starts/ends so I don't expect that tuning this could be sticked off with some simple calibration test. Your explanations are laudable but with all due respect, things behave sometimes differently as expected - short of "the proof of the pooding is in the eating".

I don't think that it should be attempted that pressure should be 0 between loops (that would demand too drastic interventions of pressure advance) - not at speeds used nowadays. I have a project planned to use a load cell sensor integrated in the filament path to actually measure it (nothing new, that has been done). If you really want to know how it goes, take just a short look here:
https://airtripper.com/1338/airtripper-extruder-filament-force-sensor-introduction/

closed 2024-07-05 by github-actions[bot]