OrcaSlicer/OrcaSlicer #4751: fan control not working as expected

lunatic1972, 2024-03-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 RC

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

K1 Max

### How to reproduce

I've attached the 3MF file.
1st thing, if "Activate Air Filtration" is NOT enabled, it's still in the code and on the first layer it turns on. If its not enabled it should never turn on and secondly I wouldn't have thought it would turn on until after "No Cooling for the first" layer, which in my case is layer 3 however it turns on at layer 1.
The other thing is that Auxiliary fan

### Actual results

Fans should not turn on until after the specified layer whether it be the print fan, exhaust fan or auxiliary fan.

### Expected results

Fans should only turn on after the specified layer. An added nice feature would be layer control for each fan

### Project file & Debug log uploads

[xyzCalibration_cube.zip](https://github.com/SoftFever/OrcaSlicer/files/14789634/xyzCalibration_cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lunatic1972, 2024-06-28:
Here's another example of the fan control being off some. Nowhere in my setup do I have fan at 100% however layer 41 and if you do a search through the gcode you will find 53 M106 S255 commands.
[Owl Planter_Tray.3mf.zip](https://github.com/user-attachments/files/16032326/Owl.Planter_Tray.3mf.zip)
[Owl Planter_Tray.3mf.zip](https://github.com/user-attachments/files/16032326/Owl.Planter_Tray.3mf.zip)

demosthenes384322, 2024-08-07:
This is still a problem. It will not turn on the chamber fan at all until after the print is complete and then it works normally. Bambu p1s

closed 2024-11-13 by github-actions[bot]