OrcaSlicer/OrcaSlicer #4752: Seam-Staggered inner seams problem

olcayoren, 2024-03-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

Macbook M1 Arm

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

On->Staggered inner seams
[image]

### Actual results

Once the gradual inseams are opened, the inseams remain aligned. Gaps are forming.

### Expected results

None

### Project file & Debug log uploads

[Uploading test2.3mf.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

olcayoren, 2024-03-30:
Hi, is this a display or printing issue? Can you comment? @igiannakas 

The situation is the same in both stable versions.
1.9.1 - 2.0.0

[image]

igiannakas, 2024-03-30:
Struggling to understand the issue. Could you give a bit more detail please?

also try it on a simple cylinder to see what the result is.

olcayoren, 2024-03-30:
> Struggling to understand the issue. Could you give a bit more detail please?
> 
> also try it on a simple cylinder to see what the result is.

The problem is seen when the inner wall gradual seam is opened. He does the stitches elsewhere. I added the video.

https://github.com/SoftFever/OrcaSlicer/assets/33813227/6d73bbac-4998-464b-be8b-194092374de8

olcayoren, 2024-03-30:
> Struggling to understand the issue. Could you give a bit more detail please?
> 
> also try it on a simple cylinder to see what the result is.

I think I misinterpreted the image. I think the seams only occur on the outer wall. When I hid the outer wall, I expected it to be hidden at the seams on the outer wall.

igiannakas, 2024-03-30:
Correct. The seam line shows if you select it on the visualiser irrespective of if the outer walls are visible or not.

olcayoren, 2024-03-30:
> Correct. The seam line shows if you select it on the visualiser irrespective of if the outer walls are visible or not.

There are no seams on the interior walls, right? Increasing the seam gap causes weakness in the inner wall. Wouldn't it be better if the inner wall seam gap was a separate area instead?

[image]

igiannakas, 2024-03-30:
Of course there are seams on the internal part as the loops need to start somewhere. To mitigate the effect (part weakness) enable staggered inner seams as you’ve done. This way the seams are offset from each other between layers by a small amount. Also enabling wipe before external wall helps a bit as it extrudes a little bit of material internally strengthening the part. Finally setting your seam gap to 0-10% at most will allow the seams to fuse together (basically tune your seam gap). 

The gaps you see in the visualiser won’t occur in real life as the hot end extrudes a tiny bit of material at the end of the wall travel, closing the seam off.

closed 2024-03-30 by olcayoren