OrcaSlicer/OrcaSlicer #4755: Z-Hop not working in Spiral Vase Mode

BK-Morpheus, 2024-03-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 RC

### Operating System (OS)

Windows

### OS Version

Windows 10 x64

### Printer

Creality K1

### How to reproduce

Slice something in Spiral Vase mode and use a few bottom layers.
Enable z-hop for all surfaces in printer / extruder settings or override it in filament settings.
The result will ignore the z-hop setting completely (no matter if set in the printer settings for the extruder, or overriding it in the filament settings).

### Actual results

During travel moves on the first layers (where solid infill is present), the nozzle scratches the solid infill. The rectraction is working correct, but the nozzle doesn't, before those travel moves (can be seen in the slicer and gcode).

If the spiral vase mode option is disabled again, the z-hop setting is working again.

### Expected results

The configured z-hop setting should even work in spiral vase mode, while printing the the configured "bottom shell layers" (where the continuous spiral move has not started yet).

### Project file & Debug log uploads

[debug_Thu_Mar_28_20_08_07_23744.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14794414/debug_Thu_Mar_28_20_08_07_23744.log.zip)
[Test_in_Vase Mode.zip](https://github.com/SoftFever/OrcaSlicer/files/14794415/Test_in_Vase.Mode.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I cross checked with version 1.9.0.0 and 2.0.0 RC. The issue is the same in both version.
In Prusa Slicer 2.7.1 the z-hop is working correctly, no matter if normal print mode or spiral vase mode.

astranberg, 2024-04-25:
I can confirm this bug.

With spiral vase on:
[image]

With spiral vase off:
[image]

[Spiral Vase z-hop Bug.zip](https://github.com/SoftFever/OrcaSlicer/files/15102929/Spiral.Vase.z-hop.Bug.zip)

Orca Slicer 2.0.0

closed 2024-08-01 by github-actions[bot]

Paubeard, 2024-10-29:
I have the exact same issue. Is there any solution for it?

Zogar89, 2025-08-02:
Same issue here
https://github.com/SoftFever/OrcaSlicer/issues/10304