OrcaSlicer/OrcaSlicer #4762: Make "monotonic line order" a separate feature the way it is made in CURA

acritum, 2024-03-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently the top/bottom infill pattern can be selected Monotonic Line or Aligned Rectilinear. I find Aligned Rectilinear infill very useful in many cases but it draws the lines in random directions and makes the final look ugly. To prevent this, Monotonic Line pattern can be used, but it doesn't care of the infill directions in the lower levels the way Aligned does.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

 I think the only reasonable way is implemented in CURA. There is a checkbox "Monotonic top/bottom order" and it can be used with ANY type of infill (concentric, line, zigzag). I think the same thing should be made in OrcaSlicer, to be able to use the advantages of both  Monotonic Line infills and any other type of infill, such as Aligned Rectilinear.

### Describe alternatives you've considered

I can't find except using CURA :)

SoftFever, 2024-03-30:
Looks like you looking for "Monotonic" pattern which is already there.
Pls pay attention, there are Monotonic and Monotonic line in the option

acritum, 2024-04-03:
It feels you don't understand what I mean. I need  Aligned Rectilinear pattern, but printed in monotonic line order. Neither  Monotonic nor Monotonic line care about the infill direction in previous top layers the way Aligned Rectilinear does, At the same time, Aligned Rectilinear cannot use monotonic infill in top layers. Please compare the images, I hope you understand what I mean. You can see that both monotonic and monotonic lines have horizontal and vertical infill in different top surfaces while alligned rectilinear has only horizontal infill (as I need), but the lines are not monotonic, they are printed partially from up to down and from down to up, and I can see the visual transition in the places they meet in the final surface.
**Aligned Rectilinear** 
[image]
**Monotonic** 
[image]
**Monotonic line**
[image]

closed 2024-07-10 by github-actions[bot]