OrcaSlicer/OrcaSlicer #4763: Variable layer height cannot specify height increase step.

acritum, 2024-03-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When using the variable layer height with Ender3-like printers, the Z-height is positioned unpredictably most of the time, the nozzle can scratch previously printed elements. This is because Orca adds a gradual unpredictable step while Ender3 Z motor accepts 0.04mm steps.

### Which printers will be beneficial to this feature?

Others, All

### Describe the solution you'd like

Make it the way CURA does. Allow to set a fixed step to increase/decrease layer height. For example, it we set it 0.04, the layers will be increased as 0.08, 0.12, 0.16, 0.20, 0.24 but not with random numbers the way it happens now.

### Describe alternatives you've considered

No alternatives at the moment, I just don't use variable layer height in Orca. Cura does it the way it should be.

vgdh, 2024-03-29:
you can configure minimum layer height in printer settings.

acritum, 2024-03-29:
> you can configure minimum layer height in printer settings.

I'm writing not about the minimum layer height but about the incremental step! It can't be configured in Orca, unfortunately.

buzzhuzz, 2024-03-29:
Out of curiosity, I've looked into the source code and see that "minimum layer height" parameter is used as a minimum layer height for adaptive slicing. At the same time, there are indeed no check/alignment for a layer height increment.

I could look into this little bit deeper, specifically because I have an Ender 3 V3 SE and I was playing with adaptive height slicing just yesterday.

vgdh, 2024-03-29:
I don't get it. Why do you want to choose increment step? What is the use case?

buzzhuzz, 2024-03-29:
I'd like to understand how things works and not sure if any changes needed at all.

Issue reporter intention to align layer height to a z-step understandable. As at first glimpse this defines our minimal increment making layers of 0.03 impossible.  However, modern stepper drivers usually supports micro stepping which makes fractional steps possible.

vgdh, 2024-03-29:
at Z axis usually are screws. And with this kind of gear reduction there are much more precision like 0.0001 available. But klipper for example uses 0.01 precision resolution for z axis because of common sense.

acritum, 2024-03-29:
When setting the layer height on your Ender 3, it is important to remember how the printer physically achieves a certain layer height. On the Ender 3, the shortest distance the printhead can move along the Z-axis is 0.04 mm, as this is equivalent to one “step” on the stepper motor. It therefore makes sense to set a layer height in a multiple of 0.04, e.g. 0.16 mm, 0.2 mm, 0.24 mm, and so forth…

For an Ender 3 with 8mm pitch lead screw and 1.8 degree stepper, you'd expect to see 8 / 200 full steps = 0.04 mm.

Yes, modern printers don't have this problem, but there are still a lot of Ender3s around because they are cheap and they give fine results for their price.

vgdh, 2024-03-29:
I think any unsupported layer height will be rounded to supported value on the printer firmware side.
But I am ot an expert on the marline, please correct me if I am wrong.

acritum, 2024-03-29:
I'm not an expert in firmware, but since I've bought Ender 3 I find it everywhere to use heights divided by 0.4. If it wasn't important I think nobody would care of it. Imagine the slicer sends requests to print at heights 0.20, 0.21, 0.22, 0.23, 0.24 so how do you think firmware should round it? if it rounds to the nearest supported by firmware, that would be 0.20, 0.20, 0.20, 0.24, 0.24, so 3 different layers will be printed at 0.20 and 2 different layers will be printed at 0.24. At a result, the wasted print in the best case, and the dead nozzle in the worst case... I think using the step supported by hardware right in the slicer is the correct solution here. CURA developers gladly understood it and implemented this solution a long time ago, but prusa/orca don't seem to care yet.

buzzhuzz, 2024-03-30:
Lets start from the hardware. 

Even original Ender 3 with A4988 drivers is configured to 16 microsteps configuration only. For the Z-axis it means that single pulse from the MCU to the motor driver will lead to:
- 1/16 = 0.0625 of a step, or a
- 1/16/200 = 0.0003125 of a revolution, or 
- Z-axis screw moves extruder 8mm per revolution. This means it moves 0.0025 per step which is true limitation.

As for the software side, Marlin has support for adaptive stepping on a capable hardware only (driver motor MS pins has to be connected to the MCU), which Ender 3s are not (however you could get it with some aftermarket board like BTT Manta E3 EZ). Because of this Z-travel is always done in 0.0025mm increments.

Marlin motion planner is there things getting really interesting:
- Marlin counting Gx inputs to an absolute position in millimeters (float data type). This includes bed mesh leveling if there are any.
- Absolute position getting multiplied by a number of steps/mm and rounded to a nearest integer. 
For example: 0.01 mm is a 4 steps. 0.011 is also 4 steps. 0.0125 = 5 steps, 0.0142 = 6 steps and so on.

All that means printer kinematics is capable to position Z-axis more accurate than Orca currently slice the model.

However, at the end all goes down to how your printer is dialed in:
- having higher than desirable resistance on Z-axis (V-wheels are to tight) means stepper need to apply higher force to move tool head. But motor torque in microstepping is usually lower than at full steps.
- having slack on Z-axis means you'll be getting not accurate position after performing Z hops.

acritum, 2024-03-30:
Mathematics is cool but the real life has its aspects :) Suppose new versions of Ender3 don't have physical limitations the way the older models did and they can lift z-axis 0.0025 mm up per step. I really doubt Ender3 can really do it because it's  not only about the stepper, but also about extremely precise geometry of the frame, extremely straight lead screw, etc. What you wrote in the last lines is 100% truth.  I've spent days to solve geometry and other problems with my Ender3 but it's just impossible due to quality issues with the hardware parts. You get what you pay for... Even the bed is 0.04-0.08 mm higher in the corners than it is in the center and I can't make it flat with the adjustment screws. It's possible to buy a new straight Z-axis stud, a new z-axis nut without a slack, a new better z-axis motor, a new straight frame,a new extremely flat bed, but I don't want to do it because finally it would be cheaper to buy a new modern printer! I'm using the printer for home hobbies only, so the quality satisfies me at the moment, although I understand that I will never get 0.0025 mm accuracy in the nozzle movements. I wonder if it can really make positioning with 0.01-0.02 mm precision...

Anyway, rounding is always a bit of mistake. If the rounding is made to a lower value, and the layer height is rather low, such as 0.08 mm, it's quite possible that the nozzle will scratch and damage previous layers sometimes. Why should we make a rounding mistake in the gcode when we can specify the exact value of the step that we want to use that will finally generate layers heights that we can predict and expect to see in the model? I don't want layers to increment to a random value such as 0.01245 mm or 0.01834 or 0.02353 or whatever the slicer decides at any moment, I want them to increment always 0.04 mm (or whatever the user specifies in the settings), this would produce predictable results and most likely reduce the possibility of partial "overwriting" of previous layers due to rounding/positioning mistakes. This is imho based on my real-life experience, I didn't make brainwashing calculations or incredible measurements with a micrometer to see the real height of layers that my printer makes, and other cool stuff. I only suggest a feature that I find reasonable and useful (as a user who sliced in CURA for 2+ years), and the final decision is made by developers of the software. Unfortunately, I'm not too smart to modify the code of the slicer to make it increment layer height in fixed steps, but if somebody can do it, I think this feature will add some value to OrcaSlicer for the owners of old and cheap hardware.

closed 2024-07-08 by github-actions[bot]

vayavalla, 2024-11-24:
I find very reasonable the request.