OrcaSlicer/OrcaSlicer #4766: Add "wall speed over overhang segments"

michael7370, 2024-03-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

ive noticed that sometimes walls over steep overhangs should be printed at lower speed because they are printed over soft and thin layer of plastic 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

my solution is to add the feature that has been described upwards 
i suggest  to add the minimum overhang degree that will be influenced (>X) and calculate the speed reduction as a percentage of the overhang speed. 

in example:
outer wall speed is  100 mm/s 
>X degrees over hang speed is 50% outer wall speed ( 50 mm/s )
 and the speed reduction settings has  been set at 150% of the >X overhang speed ( 150% of 50 mm/s is 75 mm/s ) 
this is actually means that:
100 mm/s wall speed 
50 mm/s  for >X overhangs 
75 mm/s for walls over >X overhangs

igiannakas, 2024-03-29:
Slowdown for curled overhangs slows down the internal perimeters too in areas where it might be needed (sharp edges, corners over overhangs etc).

you don’t need to slow down internal perimeters if for example you’re printing a 45 degree slope in a straight line.

michael7370, 2024-03-29:
i didnt mean the internal perimeter i mean the next layer

igiannakas, 2024-03-29:
> i didnt mean the internal perimeter i mean the next layer

Hm.. curled perimeters option does this too - it slows down higher up on the layers to mitigate the effects of curling upwards.

it would help if you shared a project image to see what you mean?

lhndo, 2024-03-29:
I've been wishing to have an option to slow down feature speeds on the layer above overhangs, bridges (including bridged infill), and supports.  
IMO this would make a considerable quality improvement, especially to the bridge scenario, where these are often  "broken" by a fast upper layer.

igiannakas, 2024-03-30:
Ah I see. It makes sense for bridges - I’ll see whether I can pull together some code for this.

lhndo, 2024-03-30:
I often print bridges at 50/70mm/s (external/internal ). The next layer on top of that is printed with regular printing speeds at 400/s or 600mm/s at 30kmm2/s and 30 SCV. 
Imagine what happens to the flimsily floating bridge lines, especially if the infill goes 90 degrees to that.

closed 2024-07-07 by github-actions[bot]