OrcaSlicer/OrcaSlicer #4786: GCODE Error, Creality CR10 SE profile

Yanrs, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality CR 10 SE

### How to reproduce

Try to slice a file using the just addedd CR 10 SE profile

### Actual results

```
Failed to generate gcode for invalid custom G-code.
machine_start_gcode Parsing error at line 7: Variable does not exist
M109 S[material_print_temperature_layer_0] 
       ^
Please check the custom G-code or use the default custom G-code.
```

### Expected results

Slice process concluding successful 

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Yanrs, 2024-03-30:
Tagging @cochcoder FYI. Pull ref. https://github.com/SoftFever/OrcaSlicer/pull/4757

Adilrafiq001, 2024-03-30:
The bed graphics are not correct and the error as reported earlier. 
[image]
[image]

As per my profile the bed texture looks like this:
[image]

Here is the Start and End Gcode that I had in my profile: 

;PRINT_START
M190 S[bed_temperature_initial_layer_single]
M109 S[nozzle_temperature_initial_layer]
M82
M220 S100 ;Reset feed rate
M221 S100 ;Reset flow rate
G28 ;Home
G92 E0 ;Reset extruder
G1 Z5.0 F3000 ;Move Z Axis up
G1 X-4 Y20 Z0.28 F5000.0 ;Move to start position
G1 X-4 Y100.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X-4 Y100.0 Z0.28 F5000.0 ;Move to side a little
G1 X-4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0  ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G1 E0.0000 F1800
;PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single]

;PRINT_END
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y220 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

cochcoder, 2024-03-30:
@Yanrs 

Can you confirm if [this](https://github.com/cochcoder/OrcaSlicer/actions/runs/8492066732) fixes it?

cochcoder, 2024-03-30:
> The bed graphics are not correctly represented and the error as reported earlier by Yanrs. ![image](https://private-user-images.githubusercontent.com/157908545/318212530-8fc5e4eb-79b2-481f-a38e-05a961ead067.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE4MjM0OTgsIm5iZiI6MTcxMTgyMzE5OCwicGF0aCI6Ii8xNTc5MDg1NDUvMzE4MjEyNTMwLThmYzVlNGViLTc5YjItNDgxZi1hMzhlLTA1YTk2MWVhZDA2Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMzMFQxODI2MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00Y2FhN2E1YmFjNGU3ZjBmYTA1ZWI3ZDM4ZDg3ZTFmZmYzODFjYjY3YWZiZGRmNzM0NjkwM2U4NDY5MzM2NTIxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5a66fxv1CqDJElcpc2_smHHcNlzI_wzHCPGVadvRy5Q) ![image](https://private-user-images.githubusercontent.com/157908545/318212579-219c80d6-9f14-4146-9436-37a910ec284a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE4MjM0OTgsIm5iZiI6MTcxMTgyMzE5OCwicGF0aCI6Ii8xNTc5MDg1NDUvMzE4MjEyNTc5LTIxOWM4MGQ2LTlmMTQtNDE0Ni05NDM2LTM3YTkxMGVjMjg0YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMzMFQxODI2MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNjMxNjhiMGYzMWYxNDcwZTExM2Y5NDhmODc5YjRjYTIxMzRlNTdkMDBhZWM0NmIyMzQzNDdjMjVkNmZmMGFiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ycaq-Eg79xsVsY6yNGBqLeCodVwDXruhMJZb9Y5kISc)
> 
> As per my profile the bed texture looks like this: ![image](https://private-user-images.githubusercontent.com/157908545/318212683-bd2dbe67-7bda-415e-a890-8253d46d088b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE4MjM0OTgsIm5iZiI6MTcxMTgyMzE5OCwicGF0aCI6Ii8xNTc5MDg1NDUvMzE4MjEyNjgzLWJkMmRiZTY3LTdiZGEtNDE1ZS1hODkwLTgyNTNkNDZkMDg4Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMzMFQxODI2MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YjIwNjRhZmMxMDE4OTlkYWNmYjNjN2Q1NjM0NTY4YjNjZTVjNTE0NWVjYjYzYjQ1MzJjM2JjNGUxNzc2MWYxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2NeL9TuIP4gWk68EfT_-Mb-mypAVoV3S38fDU1SgZvE)
> 
> Here is the Start and End Gcode that I had in my profile:
> 
> ;PRINT_START M190 S[bed_temperature_initial_layer_single] M109 S[nozzle_temperature_initial_layer] M82 M220 S100 ;Reset feed rate M221 S100 ;Reset flow rate G28 ;Home G92 E0 ;Reset extruder G1 Z5.0 F3000 ;Move Z Axis up G1 X-4 Y20 Z0.28 F5000.0 ;Move to start position G1 X-4 Y100.0 Z0.28 F1500.0 E15 ;Draw the first line G1 X-4 Y100.0 Z0.28 F5000.0 ;Move to side a little G1 X-4 Y20 Z0.28 F1500.0 E30 ;Draw the second line G92 E0 ;Reset Extruder G1 E-1.0000 F1800 ;Retract a bit G1 Z2.0 F3000 ;Move Z Axis up G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish G1 E0.0000 F1800 ;PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single]
> 
> ;PRINT_END G91 ;Relative positioning G1 E-2 F2700 ;Retract a bit G1 E-2 Z0.2 F2400 ;Retract and raise Z G1 X5 Y5 F3000 ;Wipe out G1 Z10 ;Raise Z more G90 ;Absolute positioning
> 
> G1 X0 Y220 ;Present print M106 S0 ;Turn-off fan M104 S0 ;Turn-off hotend M140 S0 ;Turn-off bed
> 
> M84 X Y E ;Disable all steppers but Z

Does [this](https://github.com/cochcoder/OrcaSlicer/actions/runs/8492559239) fix your issues?

Adilrafiq001, 2024-03-31:
This profile was created using my profile as base and I'm using my profile as of now and have provided the start and end Gcode. I have not checked it yet, but it should solve the issue. Also the bed graphics need to be solved by reading the texture abd bed stl.

cochcoder, 2024-03-31:
> This profile was created using my profile as base and I'm using my profile as of now and have provided the start and end Gcode. I have not checked it yet, but it should solve the issue. Also the bed graphics need to be solved by reading the texture abd bed stl. 

The issues should be fixed, I just want to confirm that this does fix them before it gets merged into the main program.

MARCOB3D, 2024-04-03:
The acceleration of the profile is only 500mm/s? Is that correct?

[image]

cochcoder, 2024-04-03:
> The acceleration of the profile is only 500mm/s? Is that correct? [...]

Thank you for pointing this out!

MARCOB3D, 2024-04-04:
The bed area is also wrong, the correct value is 220x220 mm

[image]

cokelid, 2024-04-17:
Looking [here](https://github.com/SoftFever/OrcaSlicer/commit/e82ea69e65ba7ee53c0a5f9a4bafb90742f9304a#diff-63ad7509e095fc1678ab59b0f1f76920e3c60948512ba3953cdd451d9b3ae15a), it still has the incorrect variable `[material_print_temperature_layer_0]`, it should be `[nozzle_temperature_initial_layer]`

liphvf, 2024-05-22:
What de diference between [first_layer_temperature]  and [nozzle_temperature_initial_layer]?

liphvf, 2024-05-22:
I don't know if it helps, but I found this configuration file in the Creality Print 4.8.2 software:

```
machine_name=CR-10 SE
machine_shape=rectangle
machine_height=265
machine_width=220
machine_depth=220
machine_heated_bed=true
machine_center_is_zero=false
has_preview_img=true
preview_img_type=png
machine_extruder_count=1
machine_head_with_fans_polygon=[[-20,10],[10,10],[10,-10],[-20,-10]]
machine_gcode_flavor=4408
machine_extruder_type=Proximal
machine_start_gcode=M220 S100 ;Reset Feedrate\nM221 S100 ;Reset Flowrate\n\nG28 ;Home\n\nG92 E0 ;Reset Extruder\nG1 Z2.0 F3000 ;Move Z Axis up\nG1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position\nM109 S[material_print_temperature_layer_0]\nG1 X10.1 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line\nG1 X10.4 Y145.0 Z0.28 F5000.0 ;Move to side a little\nG1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line\nG92 E0  ;Reset Extruder\nG1 E-1.0000 F1800 ;Retract a bit\nG1 Z2.0 F3000 ;Move Z Axis up\nG1 E0.0000 F1800 \n
machine_end_gcode=G91 ;Relative positionning\nG1 E-2 F2700 ;Retract a bit\nG1 E-2 Z0.2 F2400 ;Retract and raise Z\nG1 X5 Y5 F3000 ;Wipe out\nG1 Z3 ;Raise Z more\nG90 ;Absolute positionning\n\nG1 X2 Y218 F3000 ;Present print\nM106 S0 ;Turn-off fan\nM104 S0 ;Turn-off hotend\nM140 S0 ;Turn-off bed\n\nM84 X Y E ;Disable all steppers but Z\n
machine_max_feedrate_x=600
machine_max_feedrate_y=600
machine_max_feedrate_z=30
machine_max_feedrate_e=100
machine_max_jerk_xy=10
machine_max_jerk_z=2
machine_max_jerk_e=2.5
machine_max_acceleration_x=8000
machine_max_acceleration_y=8000
machine_max_acceleration_z=500
machine_max_acceleration_e=3000
machine_chamber_fan_exist=false
machine_cds_fan_exist=false
machine_LED_light_exist=true
```

I would like to recreate this profile in OrcaSlicer CR-10 SE Profile, but I don't know how to translate the speed settings.

[image]

PR #4793 (2024-05-22): General fixes and improvements to CR-10 SE, Ender-3 V3 KE, Ender-3 S1 Plus, & small fix for Ender-3 V3 SE
This PR is working towards fixing all the errors/bugs in the profiles that I added to Orca Slicer. 

TODO: 

- [x] Fix incorrect line width (mentioned by [exploitfate](https://github.com/exploitfate) in #4653) 
- [x] Fix incorrect line width for CR-10 SE
- [x] Fix CR-10 SE G-code error (by @bkonosky)
- [x] Add changes to CR-10 SE from @bkonosky in #4966 
- [x] Fix incorrect build plate model sizing #4786 
- [x] Fix the incorrect bed size for the Ender-3 S1 Plus 0.8 nozzle type
- [x] Make b [... cut]
closed 2024-05-22 by SoftFever