OrcaSlicer/OrcaSlicer #4787: Models off plate are visibly cut off - like draw distance

ndevenish, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0-beta (although present on earlier)

### Operating System (OS)

macOS

### OS Version

14.1 (23B74)

### Additional system information

M2 Pro. 16GB Memory.

### Printer

Bambu P1S

### How to reproduce

Add STL out of bounds. Rotate Camera.

### Actual results

When models are placed off of the plate, they are cut off by what appears initially to be Z view distance. This happens with even a few objects. Anything a certain distance off of the plate gets effectively invisible.
[image]
And even the plate control widgets appears to get cut off when at shallow angles:
[image]
However, it's not a direct view distance effect, because deliberately moving far off the plate (so it is invisible) and rotating the camera makes it appear, but appears to be cut off on the near side:
[image]

### Expected results

I expect the STL to be always visible. Here is a screenshot showing the same model and configuration (on Bambu Studio):
[image]

### Project file & Debug log uploads

[random_bin_video_cutoff.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14812501/random_bin_video_cutoff.3mf.zip)
[debug_Sat_Mar_30_15_33_12_64198.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14812505/debug_Sat_Mar_30_15_33_12_64198.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

cosmith71, 2024-04-01:
I'm seeing something very similar. Windows 11, Orca version 2.0.0 release version, RC, and Beta.
[image]

kylek29, 2024-05-30:
Yes, it's been this way since either 2.0 or 1.9 iirc. It's really prevalent on my laptop where I can't even view some models. They don't have to be off the plate, it seems to have a really small view distance.

ndevenish, 2024-05-31:
Still present in 2.1.0-beta 
[image]

ListerD000169, 2024-06-07:
Also still seeing this in 2.1 beta.  Can confirm that this was also in at least 2.0.  Running on Windows 11

closed 2024-09-13 by github-actions[bot]

ndevenish, 2024-09-13:
Definitely still here in 2.1.1

[image]

ndevenish, 2024-09-14:
And 2.2.0-beta
[image]

ndevenish, 2024-09-14:
From released binaries, this was introduced between v1.7.0 and v1.8.0-beta1.

ndevenish, 2024-09-15:
I have bisected this down and the commit causing the error is bbfb9d77fadc63ca84d70b09aa1efa6d90889736

ndevenish, 2024-09-15:
I have bisected this down and the commit causing the error is bbfb9d77fadc63ca84d70b09aa1efa6d90889736 ; this makes some sense as it does feel related to models outside the bounds of the current plate being cut off.

xsynatic, 2024-11-06:
Full 2.2.0 release

[image]