OrcaSlicer/OrcaSlicer #4788: Pressure Advance calibration is not function properly

wibiksana02, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

win 11

### Additional system information

cpu intel i7 3770

### Printer

Ender 3 Pro Klipper

### How to reproduce

GCode that generated from Pressure advance in Calibration menu, is not involving pressure advance test range, its using default value

### Actual results

GCode that generated from Pressure advance in Calibration menu, is not involving pressure advance test range, its using default value

### Expected results

GCode to test pressure advance that work

### Project file & Debug log uploads

[debug_Sat_Mar_30_22_33_00_131496log.zip](https://github.com/SoftFever/OrcaSlicer/files/14812618/debug_Sat_Mar_30_22_33_00_131496log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[Pressure Advance Test.zip](https://github.com/SoftFever/OrcaSlicer/files/14812605/Pressure.Advance.Test.zip)

vgdh, 2024-03-30:
You need to choose klipper for your machine
[image]

wibiksana02, 2024-03-30:
Hi, thank you. I forgot that marlin is not supporting PA, except recent time.

closed 2024-07-07 by github-actions[bot]