OrcaSlicer/OrcaSlicer #4789: Double Walled Vase Mode

Supahfly87, 2024-03-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Limitation of vase mode

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I am new to 3d printing and I am learning about vase mode. I like the technique but uou can't seem to make strong objects.
As I understand it, vase mode is running your print in one continuous line. I asked on reddit if there is a way to program my printer or slicer software to work as pictured in the attached image and they told me to raise a request here. Thank you for your time.
[image]

vgdh, 2024-03-30:
You will get change of the flow and that will result in poor surface finish. 
The main purpose of the vase mode - to get static flow and absent of the z-seam.
For the job you describe you can use regular print with 10% infill.

Ataraxia-Mechanica, 2024-03-31:
Try printing with 0.8/1.2 outer wall width at 50%/33% speed, And yes it is doable with just a 0.4 nozzle.

rizwansarwar, 2024-03-31:
I think you misunderstand Vase mode. It has to one continuous flow of filament to get smooth prints. From a 3d Printers perspective, it is drawing a constant straight line. For strength people use bigger nozzles and higher flow rates. 

Since the new version has scarf seam join, you can probably get really nice looking outer surface finish in normal printing mode now.

Supahfly87, 2024-04-06:
Thank you for your replies.

Supahfly87, 2024-04-06:
> You will get change of the flow and that will result in poor surface finish. The main purpose of the vase mode - to get static flow and absent of the z-seam. For the job you describe you can use regular print with 10% infill.

So as i understand it, this mode cannot do sharp corners? I liked this idea(https://www.youtube.com/watch?v=STNJbHJf9K0&t=462s) but it leaves a seam throughout the print. 
Currently i am mostly interested in it because of the lower print time. My wife does not want the printer to be on when we are not home so i cannot make larger projects without having to stop and resume several times.
Sorry if posting links is not allowed.

vgdh, 2024-04-06:
> So as i understand it, this mode cannot do sharp corners? 

Why do you think that?

Supahfly87, 2024-04-06:
is that not the change of flow you mentioned?

vgdh, 2024-04-06:
vase mode do continuous extrude and can do very sharp edges as well.
Flow change occur when extrude suddenly stop (for changing a layer or to move to another print point)

> Currently i am mostly interested in it because of the lower print time.

Your print speed is limited to the max volumetric flow of your extruder.

CCS86, 2024-04-06:
> I see. The shape I made should be printable in one line without stopping.

It's not about stopping so much as it is about a continuous progression of the Z axis. Your suggestion requires moving Z up and down to navigate that path. Your pencil evaluation is neglecting the Z axis.

Supahfly87, 2024-04-08:
The Z would not need to move down, just up

closed 2024-07-15 by github-actions[bot]

j0e57, 2025-02-14:
@JorgenAerts If you still interesting I already try to make double wall tricks for simple shape (OpenSCAD parametric), already try quite nice on orca. Better than waiting it implemented on slicer :)
  https://www.thingiverse.com/thing:6928805