OrcaSlicer/OrcaSlicer #4790: After 2.0 update Ender V3 SE 0.4 nozzle Process Drop down arrow not showing any layer height system presets. Sowing for all the other nozzle sizes/

SPYKE357, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Ender 3 V3 SE

### How to reproduce

add printer through select remove printer option
select 0.4 nozzle
go to process to select layer height system presets
none available, only shows default setting which does not match SE settings like speed/acceleration

### Actual results

no system layer height options available

### Expected results

there should be options showing for 0.12 0.16 0.2 and 0.24

### Project file & Debug log uploads

[debug_Sat_Mar_30_18_27_11_19560.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/14812688/debug_Sat_Mar_30_18_27_11_19560.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

williamcastanho, 2024-03-30:
The same is happening to me. I have already tried to reinstall after removing completely the old version, but the error persists. When I reinstalled the old version, everything came back normally.

Haleysoftware, 2024-03-30:
I found this as well and updated my user created process profiles and updated my user created printer profiles to inherit the new system profiles.

The issue is, the new printer profile "Creality Ender-3 V3 SE 0.4 nozzle" does not have any process profiles that list it as compatible. The old process profiles are "0.20mm Standard @Creality Ender3V3SE" and only list "Creality Ender-3 V3 SE" as compatible. The new ones are named for the different nozzle sizes "0.20mm Standard @Creality Ender3V3SE 0.6" and list the printer "Creality Ender-3 V3 SE 0.6 nozzle" as compatible.

It looks like the old printer profile "Creality Ender-3 V3 SE" is no longer usable or at least selectable.

A fix for this would be to update the old process profiles with 0.4 at the end and add the new 0.4 nozzle printer profile as a compatible printer.

williamcastanho, 2024-03-30:
> I found this as well and updated my user created process profiles and updated my user created printer profiles to inherit the new system profiles.
>
> The issue is, the new printer profile "Creality [...]

I will try to do this right now, thank you!

Haleysoftware, 2024-03-30:
For your user created profiles, update as below.

Filament, I did not see anything that needed to be modified in my user profile json files.

Printers, update the json file to list the correct new printer profile as "inherits"
For example, my "Creality Ender-3 V3 SE K 0.6 nozzle" I changed the inherits to "Creality Ender-3 V3 SE 0.6 nozzle". Use the new 0.2/0.4/0.6/0.8 nozzle profiles depending on what your profile uses.
The K is for Klipper in my profile name.

Process, update the json file to list the correct new process profile as "inherits". Optional, you can add the below and list the system printer profiles that you want it to show up with. I copied my profiles and made one for each nozzle size.
"compatible_printers": [
        "Creality Ender-3 V3 SE 0.4 nozzle"
    ],
For example, pretty much the same as the user printer profile above. My user process profile "Standard @Creality Ender3V3SE 0.4" still inherits "0.20mm Standard @Creality Ender3V3SE" but I had to add the compatible_printers to make it show up. My user process profile "Standard @Creality Ender3V3SE 0.6" now inherits "0.20mm Standard @Creality Ender3V3SE 0.6".

Haleysoftware, 2024-03-30:
Oh, if you are using the bambulab profile sync like I am. You will want to go to each modified profile in OrcaSlicer and do a save, even if you did no changes to any of the settings. This will update your synced files. 

I had updated my old files and just listed all the printers as compatible. I later deleted them once I created nozzle versions and the deleted files all came back when I opened OrcaSlicer but they were versions before I updated them.

SPYKE357, 2024-03-30:
im very much a noob and not sure how to go about getting the workaround done. any chance Orca Devs will fix it? id rather wait as i have a tendency of getting steps wrong and still not get the desired outcome.lol

miguelcaravantes, 2024-03-30:
same error here after the update, i think the rename of the printer is the issue
[image]

Haleysoftware, 2024-03-30:
If you want your user created process files to show up for all nozzle printer profiles, add the below to their .json files.

"compatible_printers": [
        "Creality Ender-3 V3 SE 0.2 nozzle",
        "Creality Ender-3 V3 SE 0.4 nozzle",
        "Creality Ender-3 V3 SE 0.6 nozzle",
        "Creality Ender-3 V3 SE 0.8 nozzle",
        "Creality Ender-3 V3 SE"
    ],

SPYKE357, 2024-03-30:
the added profiles for 0.2 0.6 and 0.8 nozzles are terrible, they're pretty much copy pastes of eachother, seriously lazy.  im just going to roll back to the older version. why bother for slap dash profiles. lazy work. ill just create my own. not good profiles at all, so yeh just going to roll back. why do they even bother adding it? eg 0.2 nozzle has all the same speeds, layer heights and line widths as 0.8. i mean seriously? come on Orca. tisk tisk

SPYKE357, 2024-03-30:
> If you want your user created process files to show up for all nozzle printer profiles, add the below to their .json files.
>
> "compatible_printers": [ "Creality Ender-3 V3 SE 0.2 nozzle", "Creality [...]

Thank you for your help though

Haleysoftware, 2024-03-30:
Looks like there is already a pull request #4785 to fix this issue with the system profiles. Again, you might still need to fix your user created profiles. This pull is creating 0.4 versions of all the process files.

After this update goes live, you will need to update your process files to inherit "0.20mm Standard @Creality Ender3V3SE 0.4" instead of "0.20mm Standard @Creality Ender3V3SE".  

It looks like the "0.20mm Standard @Creality Ender3V3SE" files will be gone so there will be nothing to inherit from.

Haleysoftware, 2024-03-30:
> the added profiles for 0.2 0.6 and 0.8 nozzles are terrible, they're pretty much copy pastes of eachother, seriously lazy. im just going to roll back to the older version. why bother for slap dash profiles. lazy work. ill just create my own. not good profiles at all, so yeh just going to roll back. why do they even bother adding it? eg 0.2 nozzle has all the same speeds, layer heights and line widths as 0.8. i mean seriously? come on Orca. tisk tisk
> 
> Orca literally copied one profile settings and pasted it across all nozzle sizes and layer heights, the exact same profile.... Why? What a letdown

I created my own nozzle profiles from the base one. So I guess mine all started the same but they did not stay that way for long.

SPYKE357, 2024-03-30:
> I'm sure they will fix the system profiles for the 0.4 nozzle but you might still need to update your user created profiles.
> 
> To update your files, in OrcaSlicer, go to "Show Configuration Folder" that is listed under "Help". You will find your files listed under "user" and either "default" (if you are not using the sync) or a string of numbers (It matches the sync user name). From there your profiles are split between three folders, "filament", "machine", and "process". You will need to right click the ".json" file and select a text editor. If on Windows, you can install "Notepad++" or on Windows/Mac, you can install "Visual Studio Code". From there, just modify the files as I listed above.

Thank you HayleySoftware, im just going to roll back to the beta... update isnt worth the click of a button in my opinion. not using scarf currently so update is wasted on me, if these profiles were proper profiles then i defintely would have stayed on the update. the nozzle size profiles are pretty dumb and pointless. the same copy pasta template for all layer heights and nozzle sizes is really pointless. the amount of poor quality/or even worse/failed prints/clicking extruders these profiles will cause is just shameful...really low effort

williamcastanho, 2024-03-30:
> If you want your user created process files to show up for all nozzle printer profiles, add the below to their .json files.
>
> "compatible_printers": [ "Creality Ender-3 V3 SE 0.2 nozzle", "Creality [...]

For those who maybe don't know where to change it:

1. Press win + r or search in windows start for: "%appdata%" without ".

2. It will open an address like: C:\Users\your nome\AppData\Roaming

3. Find the past called OrcaSlicer, then go to system --> Creality --> process

4. Find the profiles as: 0.12mm Fine @Creality Ender3V3SE.json

5. Open it with the note. If you need it you probably don't have notepad++ or similar. So right-click and choose open with, then pick out Notepad.

6. Go to the bottom of the document and add what @Haleysoftware instruct:

[
		"Creality Ender-3 V3 SE 0.2 nozzle",
		"Creality Ender-3 V3 SE 0.4 nozzle",
		"Creality Ender-3 V3 SE 0.6 nozzle",
		"Creality Ender-3 V3 SE 0.8 nozzle",
		"Creality Ender-3 V3 SE"
	]

7. You will need to add this for every profile.

8. Restart Orcaslicer.

jaeseopark, 2024-04-02:
I was able to fix the issue by following the solution posted by @Haleysoftware Thanks!!!

wodji, 2024-04-02:
https://github.com/SoftFever/OrcaSlicer/discussions/4828

NebaDrago27, 2024-05-28:
Hi 
My 3d printer have same problem!!!!
How I can fixit???
😭😭😭😭

closed 2024-10-21 by github-actions[bot]