OrcaSlicer/OrcaSlicer #4792: In the new version, my profile disappears/appears/disappears.

ZinovievOleg, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

Catalina 10.15.7

### Additional system information

intel core i5, 8 ggb, intel iris pro 1536 mb

### Printer

AnyCubic kobra 2 pro

### How to reproduce

After upgrading to version 2.0.0, I found that my previously saved settings profiles disappear and only the system settings remain.
[image]

### Actual results

I tried to restore my settings via import, but they did not appear. And I get an error ("0 configurations have been imported (only non-system and compatible). Hint: Before importing the configuration, make sure that you have added the appropriate printer"). After several "OrcaSlicer" reboots, my profile reappeared. And so several times, when the program is opened again, my profile is not there, but after a few reboots of the slicer, the profile appears by itself. If you install the old version 1.9.1, this problem does not occur.
[image]
[image]

### Expected results

It is desirable that the profile does not disappear. And it's a huge plus if, when you open the program again, the active profile was the one that was previously selected.

### Project file & Debug log uploads

[debug_network_Sat_Mar_30_21_02_15.log.enc.zip](https://github.com/SoftFever/OrcaSlicer/files/14812907/debug_network_Sat_Mar_30_21_02_15.log.enc.zip)
[pen-box-louvre-v4.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14812908/pen-box-louvre-v4.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I really hope for your help.
And I want to thank you for the best slicer in the world! My old Mac system can afford the latest version of the slicer. It is very fast and has fantastic features. Other slicers are in second and third place! You are the best guys!

ZinovievOleg, 2024-03-30:
You are the best!

Achduka, 2024-03-30:
Have the same problem

TheMeanCanEHdian, 2024-03-30:
Also experiencing this. Happy to provide any files that might be needed to help solve.

jamzilla, 2024-03-31:
Also happening to me in both the beta and stable version of 2.0.0. Open Orcaslicer and the custom profiles are missing, close and re-open Orcaslicer and the custom profiles are there, close and re-open Orcaslicer and the profiles are missing again... just keeps repeating. 

Kinda annoying and haven't found a fix yet. 

Tried uninstalling and deleting folder in %AppData% and clearing any registry files for Orcaslicer to no effect. The "Auto sync user presets (Printer/Filament/Process)" also has no effect.

Looking at some of the other issues posted and this appears to be a duplicate of around 3-4 posts with the same problem. 

Anyone have a hotfix for this issue?

ZinovievOleg, 2024-03-31:
> Со мной также происходит как в бета-версии, так и в стабильной версии 2.0.0. Откройте Orcaslicer, и пользовательские профили отсутствуют, закройте и повторно откройте Orcaslicer, и пользовательские профили есть, закройте и повторно откройте Orcaslicer, и профили снова отсутствуют... просто продолжает повторяться.
> 
> Немного раздражает, и пока не нашел исправления.
> 
> Пытался деинсталлировать и удалить папку в %AppData% и очистить все файлы реестра для Orcaslicer, но безрезультатно. "Настройки автоматической синхронизации пользователя (принтер / Нить накала / процесс)" также не дали эффекта.
> 
> Просматриваю некоторые другие опубликованные сообщения о проблемах, и это, похоже, дублирует примерно 3-4 сообщения с той же проблемой.
> 
> У кого-нибудь есть исправление для этой проблемы?

Dear Jamzilla, I have a Mac system and the error is exactly like yours. I think there is a common mistake here. I can say that if the error is massive, then dear developers will definitely pay attention to it and help.

ZinovievOleg, 2024-03-31:
> Also happening to me in both the beta and stable version of 2.0.0. 
> 

Dear Jamzilla, I have a Mac system and the error is exactly like yours. I think there is a common mistake here. I can say that if the error is massive, then dear developers will definitely pay attention to it and help.

tferre61, 2024-03-31:
It happens to me too, I installed the new version 2.0, to recognize the imported profile, I have to close and reopen orca every time. WINDOWS 11

jmvaughan2, 2024-04-03:
Same here. Beta and RC.   #4459 is one of them from a few weeks ago.

ZinovievOleg, 2024-04-04:
I installed [2.1.0](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds) beta - the problem remains, the profile (backup) is not imported. I'm staying on the best 1.9.1 version at the moment.

ZinovievOleg, 2024-04-24:
upd: I didn't wait for the fix and returned to prusa. =))

jmvaughan2, 2024-06-15:
> I installed [2.1.0](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds) beta - the problem remains, the profile (backup) is not imported. I'm staying on the best 1.9.1 version at the moment.

Also experiencing issue still in 2.1.0-Beta

jaanshgo, 2024-09-10:
Same problem with 2.1.1. My bandaid solution is to save projects with my profiles and restore them when opening the project. Exporting profiles doesn't work for me either so I do it on projects. Apple M1 pro Sonoma 14.0. I really don't want to switch to prusa slicer but It's looking like I'll have no choice. Restoring the profiles multiple times a day gets time consuming and tiring.

TheMeanCanEHdian, 2024-09-10:
Rebooting after updating solved my issue with profiles disappearing.

closed 2024-12-18 by github-actions[bot]