OrcaSlicer/OrcaSlicer #4794: Small area flow compensation crash to desktop.

PhilBaz, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 release 

### Operating System (OS)

Windows

### OS Version

windows 10

### Additional system information

AMD 2900 wx 1070ti

### Printer

Voron 2.4

### How to reproduce

Add primitive one cube, one cylinder. Using my normal print process for ABS, only change is turn on new small area flow comp. Slice, slicing hangs, crash to desktop.

### Actual results

Crash to desktop.

### Expected results

Slice file.

### Project file & Debug log uploads

[OC2.0crash.zip](https://github.com/SoftFever/OrcaSlicer/files/14813072/OC2.0crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Orca has quickly become the best slicer. Thanks you! Also, please look at the brim doesn't follow elephants foot comp. bug i posted, that's a big one. But mostly thank you!

Hlatbern90, 2024-04-01:
I am having this issue on Linux (Mint 20.3). No crash files are generated and the program closes. Models slice just fine with this feature turned off.

IguanaBob, 2024-06-27:
I also see this behavior with Xubuntu 20.04.4 LTS.

Running the AppImage from the command line I see only this output between hitting slice button and crashing to desktop.

Segmentation fault (core dumped)

No output in the log directory.

I'm not sure how to gather better info but please let me know if there is something more I can provide.

closed 2024-10-05 by github-actions[bot]