OrcaSlicer/OrcaSlicer #4795: Missing Printers

ThadudeIntheCorner, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Most Creality printers

### How to reproduce

Go to the "Printer Selection" tab and scroll down to the Creality section.

### Actual results

There are only two printers in the Creality section: the CR-10 V2 and the CR-10 Max. 
[image]

### Expected results

I'd like to have an option for my printer (Ender 5 S1) so that I can actually use it.

### Project file & Debug log uploads

Nothing to add.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TexZeTech, 2024-03-31:
I just checked on my windows vm and linux vm, this isn't an issue I see.

I would recommend at the minimum uploading your log files and try the check for updates button?

[image]

Beyond that I don't know.

ThadudeIntheCorner, 2024-03-31:
It's the newest version. When I updated to V2.0.0, that's when my printer disapeared.

ThadudeIntheCorner, 2024-04-03:
Redownloaded it, everything was there. Wierd.

closed 2024-04-03 by ThadudeIntheCorner