OrcaSlicer/OrcaSlicer #4796: Cannot select printer when using create nozzle from existing printer

Nadanoob99, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

21H2

### Printer

All printers

### How to reproduce

1. Click on printer 2. Click on create printer 3. Select create nozzle for existing printer 4. click the dropdown menu to select printer.

### Actual results

When I select the dropdown menu it flips from white to green and that's it.

### Expected results

A dropdown with my printers should appear.
[image]

### Project file & Debug log uploads

[Untitled.zip](https://github.com/SoftFever/OrcaSlicer/files/14813255/Untitled.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

monyarm, 2024-04-08:
Same issue here on Gentoo Linux with Orca Slicer 2.0.0.

craftguy, 2024-05-31:
Same problem here on 2.1.0 beta

Yasamal33, 2024-07-31:
Same issue on 2.1.1

St0nefish, 2024-08-24:
seeing the same issue on Nobara Linux using version 2.1.1

JulienRee, 2024-09-24:
This is still an issue! How is this not solved yet?

navi21, 2024-11-24:
Still not solved on 2.2.0 after more than half a year!!

B-Man1, 2024-12-18:
I also have this issue

closed 2025-04-29 by github-actions[bot]