OrcaSlicer/OrcaSlicer #4798: Anti virus program deletes current 2.0 version

Skyl-Orca, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambulab X1

### How to reproduce

Unfortunately, my antivirus program still deletes the exe of the current stable version 2.0 :-( something still seems to be recognized as a "threat".

### Actual results

[image]

### Expected results

-

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TexZeTech, 2024-03-31:
Whats the Anti-virus?

Skyl-Orca, 2024-03-31:
Norton

TexZeTech, 2024-04-01:
If you are truly concerned email NortonAV and get them to check the file.

I would say it's most likely a false positive and I would just hit ignore, however if you are even slightly concerned that your file has been compromised check the SHA256 of OrcaSlicer_Windows_Installer_V2.0.0.exe and verify it.

I just downloaded the exe from https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.0.0 .

SHA256: `58b400ec976d1e38f8876d75b6e7c6ac9945cb5f231705b71c96d10e474ad32c`

[image]

[VirusTotal](https://www.virustotal.com/gui/file/58b400ec976d1e38f8876d75b6e7c6ac9945cb5f231705b71c96d10e474ad32c) says its clean.

[image]

justanotheruser4once, 2024-06-05:
Aside from Norton, there are 2 additional AV engines that mark the Installer (but not the portable version) as malicious. These are the Bkav (report from VirusTotal) and the HybridAnalysis (CrowdStrike Falcon sandbox), and the hash is correct - 58b400ec976d1e38f8876d75b6e7c6ac9945cb5f231705b71c96d10e474ad32c. I've seen this a fair amount with installers specifically. There is a currated list of contacts for the different AV vendors - [AV False Positive Help](https://github.com/yaronelh/False-Positive-Center)

Bkav / VirusTotal Report
- False Positive Reports Contact: [fpreport@bkav.com](mailto:fpreport@bkav.com), [bkav@bkav.com](mailto:bkav@bkav.com)
- https://www.virustotal.com/gui/file/58b400ec976d1e38f8876d75b6e7c6ac9945cb5f231705b71c96d10e474ad32c
[image]

CrowdStrike Falcon / HybridAnalysis Report
- False Positive Reports Contact: [VTscanner@crowdstrike.com](mailto:VTscanner@crowdstrike.com)
- https://hybrid-analysis.com/sample/58b400ec976d1e38f8876d75b6e7c6ac9945cb5f231705b71c96d10e474ad32c/660c58399878e649e9000fe9
[image]

Specifically for Norton, they have provided a link:
[Norton FP Submissions](https://submit.norton.com/)

Anyone can reach out, but I think FP submissions would be better received from developers that may be able to answer any question that arises.

liamwerk7676, 2024-08-05:
for me it got 8 / 10 malware score https://tria.ge/240805-tpk5wa1fjp/behavioral2 i dont know if i can trust it 
[image]

liamwerk7676, 2024-08-06:
update: i think its safe i am using it rn and i havent yet had any isues and malware detection on my pc didnt go VIRUS !!! and delted it so i think its safe

TexZeTech, 2024-08-08:
> for me it got 8 / 10 malware score https://tria.ge/240805-tpk5wa1fjp/behavioral2 i dont know if i can trust it [...]

Why even comment on this? You checked the 2.1.1 version not the 2.0 version. 

That's like someone talking about a home that was burnt down the previous year then you come back and say "Yep looks like there is no fire here." after the home has been remade.

I request this ticket gets closed.

justanotheruser4once, 2024-08-09:
> Why even comment on this? You checked the 2.1.1 version not the 2.0 version.
>
> That's like someone talking about a home that was burnt down the previous year then you come back and say "Yep looks [...]

Although I agree, it would have been better for the user to report the issue with 2.1.1, adding it to an open issue for 2.0 is not useless. It helps other users who are trying to make an informed risk accessment about using the software. Just because you don't value security to the same extect of others, does not make the issue any less relevant to the project.

Version 2.0 is the latest release that has been digitally signed, and people that value using FOSS while trying to address possible supply chain concerns depend of this. Additionally, the 2.1.1 installer is also identified malicious on VirusTotal so upgrading to it does not resolve the issue.

The issue is not with the project itself, but the installer. Reviewing Hybrid Analysis reports lead me to believe it's likely a false positive just due to the nature of installers. However, reviewing the log analysis of tria.ge makes me question my initial assumption. The developers have not given a response to indicate they checked to verify the installer was not tampered with before distribution, neither for 2.0 nor 2.1.1. Since 2.0 was digitally signed, I would like to think the code changes were reviewed before being signed, but I would just like confirmation what was used to create the installer was not tampered with. 

If I had the capability to compile from source, I would be able to confirm it either way. So really we are asking for comfirmation that the installer was verified as not being tampered with, but working with AV vendors to get the matter resolved would solve the problem for future releases.

closed 2024-11-15 by github-actions[bot]