OrcaSlicer/OrcaSlicer #4801: After update from beta to official 2.0.0 Settings are gone

LarsOde, 2024-03-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

Ender-3 V3 SE

### How to reproduce

open new project
in settings, there's only "Default Settings" those made in previous versions are not in dropdown

### Actual results

previews settings missing, i.e. all Line width settings are set to "0"

### Expected results

previous settings taken over from previous software version

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14813419/log.zip)
[D3D Ender 3 V3 KE Y Rails MGN12H_send.zip](https://github.com/SoftFever/OrcaSlicer/files/14813422/D3D.Ender.3.V3.KE.Y.Rails.MGN12H_send.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

buckeyestargazer, 2024-03-30:
Me too.  All of my user printer, filament, print profiles are gone.  Please tell me that I can recover them...

melkosofter, 2024-03-31:
> Me too. All of my user printer, filament, print profiles are gone. Please tell me that I can recover them...

same issue. found a solution - just open another instance of Orca (don't close first) and you will see all profiles

LarsOde, 2024-03-31:
> > Me too. All of my user printer, filament, print profiles are gone. Please tell me that I can recover them...
> 
> same issue. found a solution - just open another instance of Orca (don't close first) and you will see all profiles

can't confirm... Printer & filament profiles are there (but were before) but no settings profiles

buckeyestargazer, 2024-03-31:
> > Me too. All of my user printer, filament, print profiles are gone. Please tell me that I can recover them...
> 
> same issue. found a solution - just open another instance of Orca (don't close first) and you will see all profiles

That didn't work for me. Thankfully I had previously exported all settings, but that was a few months ago.  However I had also exported sliced plate files and one way to recover the settings is to open a previously exported .3mf which will bring the settings in, then just save the print/filament/printer settings.

LarsOde, 2024-03-31:
> That didn't work for me. Thankfully I had previously exported all settings, but that was a few months ago. However I had also exported sliced plate files and one way to recover the settings is to [...]

tried that, opened a .3mf, saved the settings from there, opened a new project, settings not available... my guess is it's something simple as paths

connordeckers, 2024-03-31:
The commit that broke this (for my Creality Ender3 V3SE), is here: https://github.com/SoftFever/OrcaSlicer/commit/a2d24cd67ff242a710c3e9c1f758ea2870400c68

I managed to get my processes to reappear by adding the adding to the `compatible_printers` array for each `system/Creality/(profile) @Creality Ender3V3SE.json` file:

```diff
	"compatible_printers": [
-		"Creality Ender-3 V3 SE"
+		"Creality Ender-3 V3 SE",
+		"Creality Ender-3 V3 SE 0.4 nozzle"
	]
```

I know this isn't a permanent fix, but it at least lets me get on with printing again!

buzzhuzz, 2024-03-31:
There are PR #4785 that fixes the problem

jmvaughan2, 2024-06-15:
this is still present in 2.1.0-Beta for non-Creality printers that PR #4785 addressed

closed 2024-09-21 by github-actions[bot]