OrcaSlicer/OrcaSlicer #4804: layer_num variable starts at 0 instead of 1

JCBond, 2024-03-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Voron 2.4

### How to reproduce

Use the layer_num variable in a custom g-code

### Actual results

layer_num starts at layer 0

### Expected results

layer_num should start at layer 1

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

foreachthing, 2024-03-31:
I agree.
But please don't just change it without notice. I have post processing scripts replying on that number.

JCBond, 2024-03-31:
> Uhm, why the double post? #4802
Sorry for the double post. I initially thought of putting it only on the discussion side, but then figured it was actually an issue.

Klober81, 2024-04-30:
Unfortunately, layer_num actually starts at -1 if used in "Change filament G-code" of the Machine G-code in the Printer settings. It would make more sense if the same starting number is used for all portions of the first layer regardless of which custom G-code section it's being used in. The following screenshot is an example of the result of using ";Layer#=[layer_num] - Z=[layer_z]" in "Change filament G-code".
[image]

closed 2024-08-07 by github-actions[bot]