OrcaSlicer/OrcaSlicer #4808: not support --datadir

gcastle2007, 2024-03-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: Ryzen 5 3600
Memory: 16Gb

### Printer

Voron 2.4

### How to reproduce

I use portable version 1.9.1 with separate config folder and have start command:
D:\portable\OrcaSlicer_Windows_V1.9.1_portable\orca-slicer.exe --datadir="d:\portable\OrcaSlicer_Windows_V1.9.1_portable\conf"

I download portable version 2.0.0 and extract.
I copy folder with config to 2.0.0 and set --datadir for start:
D:\portable\OrcaSlicer_Windows_V2.0.0_portable\orca-slicer.exe --datadir="D:\portable\OrcaSlicer_Windows_V2.0.0_portable\conf"

### Actual results

Orca not pickup config from --datadir.

### Expected results

Orca pickup config from --datadir and offers to configure the configuration from scratch.

### Project file & Debug log uploads

debug_Sun_Mar_31_10_06_15_3220.log.0

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[debug_Sun_Mar_31_10_06_15_3220.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14814467/debug_Sun_Mar_31_10_06_15_3220.log.zip)

TexZeTech, 2024-03-31:
So I tried to replicate this on my windows 10 box but not on a secondary drive just in the command prompt.

It seems to work fine

My first go I just let it generate the configuration directory.

`C:\Users\Admin\Desktop\OrcaSlicer-portable\OrcaSlicer_Windows_V2.0.0_portable\OrcaSlicer\orca-slicer.exe --datadir="C:\Users\Admin\Desktop\OrcaSlicer-portable\OrcaSlicer_Windows_V2.0.0_portable\OrcaSlicer\conf"`

My second go I copied the configuration folders from `C:\Users\Admin\AppData\Roaming\OrcaSlicer` into conf2

I then ran ``C:\Users\Admin\Desktop\OrcaSlicer-portable\OrcaSlicer_Windows_V2.0.0_portable\OrcaSlicer\orca-slicer.exe --datadir="C:\Users\Admin\Desktop\OrcaSlicer-portable\OrcaSlicer_Windows_V2.0.0_portable\OrcaSlicer\conf2"``

It worked as expected.

gcastle2007, 2024-03-31:
Have you checked that the folder content is created when starting from scratch. 

It is start Orca 2.0.0 after copy conf folder from 1.9.1:
[image]

TexZeTech, 2024-03-31:
> Have you checked that the folder content is created when starting from scratch.
> 

No I didn't read your log file and from what I'm reading it wouldn't have helped me. I tried to replicate it from scratch by downloading and setting a directory and copying my configs from the app directory to said new directory and I could not replicate it. I then responded with my results to you.

I'm just trying to narrow down what is happening.

It could be exactly what you are saying, that it has issue from referencing a secondary drive that isn't the main drive. eg; drive D: 

To start with can you please put code/log text in `code blocks` so we don't ping users unintentionally and lets others copy the text as you have it entered?

Next can you upload your configs directory so others can replicate what is happening with the files.

Can you also try just extracting OrcaSlicer to a folder on the desktop and replicating what I've done to see it does the same thing using command prompt?

`%USERPROFILE%\Desktop\OrcaSlicer-portable\OrcaSlicer_Windows_V2.0.0_portable\OrcaSlicer\orca-slicer.exe --datadir="%USERPROFILE%\Desktop\OrcaSlicer-portable\OrcaSlicer_Windows_V2.0.0_portable\OrcaSlicer\conf"`

closed 2024-07-08 by github-actions[bot]