OrcaSlicer/OrcaSlicer #4809: Orca crashes when trying to create custom filament profile.

Berkano24, 2024-03-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10 / Windows 11

### Printer

Flashforge Adventurer 5M Pro

### How to reproduce

When trying to create a new custom Filament profile, the slicer immediately crashes as soon as you select the reference Filament profile to copy from.

### Actual results

Slicer crash  

### Expected results

It should be possible to to create a new filament profile.

### Project file & Debug log uploads

[crash_Sun_Mar_31_10_32_53_0.log](https://github.com/SoftFever/OrcaSlicer/files/14814699/crash_Sun_Mar_31_10_32_53_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

brandonjkessler, 2024-04-10:
I'm also suffering the same crash on OrcaSLicer 2.0.0, On Windows 11 23H2, and Flashforge Adventurer 5M (Non-Pro).

Error:
[image]

Example Before Crash:
[image]

I also attempted using the AnkerMake M5, But the only PLA showing up in the Filament Preset is the FlashForge Generic PLA still, even with all the other options listed.

After removing all references to the FlashForge and only having the AnkerMake M5 listed, I was able to add Custom Filament.

Once I removed AnkerMake M5 and re-added FlashForge, problem happened again. Select "Create Based on Current Filament" and "Copy Current Filament Preset" causes a crash.

MadCatUSA, 2024-04-12:
Same issue as @brandonjkessler. Exactly the same issue. Flashforge Adventurer 5, trying to create a custom filament profile the software crashes.

a0s, 2024-04-16:
Same issue on mac m2. It happens when i try to save User Preset (for Process) with custom name. 
It looks like it can't handle spaces in the profile name (like it can't create a folder or something)
Possible temporary solution: Save the user profile with a name that does not contain spaces. Then you can delete it and after that you can give any names to the profiles.

SimonSolar2C, 2024-05-11:
I am also getting similar crash - program locks up when I save current modified profile with a new name, and responds with this screen when clicking on it. Waiting stays opaque until I force close the app. Version is 1.9.1
[image]

timnolte, 2024-05-14:
I'm also having the same problem, FlashForge Adventurer 5M, I'm using the latest v2.0.0 Linux AppImage.

timnolte, 2024-05-14:
This appears to probably be a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/3242

closed 2024-08-21 by github-actions[bot]