OrcaSlicer/OrcaSlicer #4811: Customized Filaments under devices

marcuswitt, 2024-03-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambulab P1S

### How to reproduce

In version 1.9.1 no problem to select my filaments under Devices

[image]
[image]

### Actual results

I have this display error in version 2.0.0. The red border is normally red and the pen for editing is missing. 
Furthermore, I cannot select my filaments in this version as I can in version 1.9.1. 

[image]

### Expected results

..

### Project file & Debug log uploads

[debug_network_Sun_Mar_31_11_22_40.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14814908/debug_network_Sun_Mar_31_11_22_40.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

What is very strange is that when I do a resync under prepare my filaments are synchronized, but as written, I cannot select them under devices in the ams. there are only the standard filaments.

skvarel, 2024-03-31:
Der Stift ist noch da. Man sieht ihn nur nicht, da er auch Weiß ist. Ich habe hier das gleiche Problem.

TexZeTech, 2024-03-31:
Going off what skvarel said: google translation: "The pen is still there. You just can't see him because he's also white. I have the same problem here.".

I think what needs to happen is an outline that is black or white of said pen or a color change of the pen depending on the colour. 

@marcuswitt can you confirm if the pen is there?

marcuswitt, 2024-04-02:
> Going off what skvarel said: google translation: "The pen is still there. You just can't see him because he's also white. I have the same problem here.".
>
> I think what needs to happen is an outline [...]

[image]

i'm not sure if there is a pen there

marcuswitt, 2024-04-02:
i changed the color from red to green to test.... that worked and it was then displayed completely green and the pen is also visible. but the problem is that i still can't select my custom filaments

ichisaur, 2024-04-02:
I have the same problem here. No issues with the edit button color, but the custom filament options were working in 1.9.1, completely gone in 2.0.0. 

I can still select them via the handy app, just not OrcaSlicer. Bambu Slicer does not show the custom filaments either, though I'm not sure if it ever did.

marcuswitt, 2024-04-04:
i have just installed it on a completely new pc as a test, unfortunately exactly the same that it is not displayed correctly and i can't select my custom filaments in the ams settings either

Crossfire1189, 2024-04-04:
same problem with custom filaments here. worked in v1.9.1.

jeremydthomas, 2024-04-05:
I also cant select custom filaments in the ams device tab, I never could actually

marcuswitt, 2024-04-11:
i have opened a ticket with bambu the last few days because the user-defined filaments cannot be selected in the studio in the ams because it works with the P and A. Bambu apparently has a "bug" and wants to fix this problem with the next firmware. maybe this is also the reason 
i have added my filaments in the ams with version 19.1 and opened the 2.0.0 version on another device, synchronized and noticed that the filaments are apparently entered in the ams, but are not displayed in the selection menu. 

if you do a "resync" in the 2.0.0, they are read out correctly from the ams, as written only the display itself is missing in the selection list in the ams

[image]

ichisaur, 2024-04-22:
What firmware version is this and would you mind posting the notes? I’m in the beta program but I don’t see a pending update.

marcuswitt, 2024-04-23:
[image]
[image]

closed 2024-07-30 by github-actions[bot]