OrcaSlicer/OrcaSlicer #4819: Layer number is not shown (-)

witchcraftsman, 2024-03-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

mac os 12.7.4

### Printer

Qidi x-max 3

### How to reproduce

1. Load project
2. Preview/Slice
3. Start printing 
4. Switch to device tab

### Actual results

Current printed layer shown as '-' (dash)

### Expected results

Current layer number should be shown instead in device tab

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14817144/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

The issue was actually with previous major version of the app.
NOTE: The workaround is to close OrcaSlicer and start again, layer number gets shown.

TexZeTech, 2024-03-31:
Add a screen shot please?

you can also resize the image in question with `<img src="image.jpg" width=200px height=200px>` or `<img src="image.jpg" width=80% height=80%>`

[image]

witchcraftsman, 2024-04-17:
[
[image]
](url)

witchcraftsman, 2024-04-17:
And finish time calculation (which is also wrong) is possibly related

closed 2024-07-24 by github-actions[bot]