OrcaSlicer/OrcaSlicer #4833: Overhang Speed not reducing correctly

mbrady4597, 2024-04-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

V-core 3

### How to reproduce

Slice a model with flat faces at an angle with slow down for overhang speeds set below the outer wall speed.

### Actual results

The speed for outer walls changes on some of the overhangs but not all. The first picture shows the new slow down method. You can see the walls are slowing down toward the bottom but higher on the model, there is no slow down for the overhangs.
[image]
In the second picture, the only setting changes was turning on classic mode. Here, all overhanging outer walls have the appropriate slow down. 
[image]
Slow down for curled perimeters does not change this problem and I have found it occurs at different angle as well if I slightly rotate the model.

### Expected results

Overhanging walls should slow down according to the overhang %. I have an outer wall line width of .56mm and layer height of .2mm. The model's overhang is 65 degrees from horizontal. This calculates to .09mm of overhang and 16% of line width, so the speed should be adjusted to 100mm/s for the 10-25% range. It just doesn't make sense why some of the walls have been slowed but others haven't that have the same overhang angle.

### Project file & Debug log uploads

[Overhang Piece.zip](https://github.com/SoftFever/OrcaSlicer/files/14822681/Overhang.Piece.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

vgdh, 2024-04-01:
yeah that is weird.

mbrady4597, 2024-04-01:
I thought maybe the new mode was using interpolation, just couldn't find that info in any documentation when searching. It seems there are still some other oddities though. I figured out why the slowdown was only happening in the lower part of the model and it relates to extrusion rate smoothing, which I believe is another bug itself.  So in these slices, I turned the extrusion rate smoothing to 0. I lowered the overhang speed in the 10-50% ranges to exaggerate the effects too.
[image]
Here, using classic mode, the overhang is slowed down because it is in the 10-25% range.
[image]
Here, with classic mode disabled, there is no slow down for the front face overhang, despite still being in the 10-25% range. Even if interpolation is happening, there should be some slow down. Testing further, I didn't get any overhang slowdown until I rotated the model a full extra 10 degrees. This leads me to believe the new mode calculates overhang % in a different (incorrect) way.

mbrady4597, 2024-04-01:
And the separate issue (can start a separate issue if required) is that extrusion rate smoothing seems to hold velocity low between overhangs in the same layer.

Expected behavior: Extrusion rate smoothing smooths the slowdown and speed-up going into and out of an overhang that has reduced speed compared to the outer wall speed. After an overhang, the outer wall should pick back up to the outer wall speed.

Actual result: With extrusion rate smoothing on and regardless of the rate (tested 20-20000), the outer wall between two distant overhangs is kept at the overhang speed instead of speeding up to outer wall speed then slowing back down for the next overhang. This was evident in the slicer and was noticeable when printing too. It resulted in over-cooling of the outer wall on the long straight section due to the slower resulting speed.
[image]
Here with extrusion rate smoothing off.
[image]
And with it on.
[image]
Here you can see it happening on parts of the back of the model too. Interesting that it only happens when the overhangs are connected by a straight line. So it happens between the two edge overhangs on the front side, but not on the back side where the brace sticks out. Maybe the issue is it's not breaking the long segment into smaller chunks like it's supposed to. As in it can't ramp up/down speed in a single gcode command for a long line. The line must be broken up into many segments for this to happen. I thought this was how extrusion rate smoothing/smoothing segment length worked, but it doesn't seem to cover this case.

igiannakas, 2024-04-01:
Can you post the project file for the above please?

mbrady4597, 2024-04-01:
[Classic no smoothing.zip](https://github.com/SoftFever/OrcaSlicer/files/14827989/Classic.no.smoothing.zip)
[No classic with extrusion smoothing.zip](https://github.com/SoftFever/OrcaSlicer/files/14827990/No.classic.with.extrusion.smoothing.zip)

Manu512, 2024-05-23:
I have a similar problem. It seems that the speed for overhangs between 75% to 100% is reversed with the speed of the bridge from what I have observed

evilDave, 2024-07-12:
It's very weird.
From what I have tested by slicing a file with 0% - 100% overhangs is this:
- the interpolation is done between start and end overhang values, from the previous speed, up to the speed configured for that "bracket". which is certainly not what I had assumed it would be (my guess was of the value to be used at the center of the range and it to be interpolated both ways to the previous and next values), but nonetheless it makes sense.
- there is no interpolation for 10-25
- the upper limit for the 75-99 range is wrong (88)

```
outer wall: 100
10-25 set to 80
25-50 set to 60
50-75 set to 40
75-100 set to 20
bridge set to 100

the interpolation will look like:

0-9% = 100
10-24% = 100
25-49% = 80...60
50-74% = 60...40
75-87% = 40...20
88-100% = 100 bridge speed
```

this is bourne out by slicing a cube with a cylinder punched through it (0.2mm layer height, 0.4mm line width):
[image]

looking into the gcode:

at the first speed adjusted layer there is G1 F4787.822
which is 79.9mm/s
x pos is 73.574 and was 73.473
diff of 0.101 which is 25.25%
so it has switched straight from 100 to 80mm/s at 25% rather than interpolate 100 to 80 from 10% to 25%

at the last layer before switching to 100mm/s is G1 F1206.594
which is 20.1mm/s
X pos is 91.346 and was 90.999 on the previous layer
difference of 0.347 /0.4 nozzle is 86.75%
so it has interpolated down to 20mm/s but only to 87% rather than all the way to just under 100%

NB: unless there is some mistake I am making about those calculations? It looks pretty cut and dried from the image anyway, the rainbow should go all the way from red to blue but skips orange. And I am pretty sure that's the right way to figure out the overhang percentage...

-- 

I had a go looking at the source and had to give up trying to find exactly where the overhang calculation is done, or where the interpolation is done.

The result though is I am able to now enter values into the table in a way that makes sense, and I know that I MUST use support threshold angle about 30 otherwise at 60deg it will have slowed down from 80 to 20, and then when it hits 61deg it will jump back up to the bridge speed of 80! (which pulls the corner all the way around) It's a shame because with the slowdown turned off and the normal speed turned down to a suitable 10 I can print an overhang of all the way to almost 95.

igiannakas, 2024-07-12:
I believe less than 5-10% overlap is considered a bridge hence it prints with that speed. Enabling slowdown for curled overhangs overrides this and uses the declared speeds for the perimeters.

evilDave, 2024-07-14:
That pretty much matches with my findings, less than 13% overlap will be treated as a bridge. I feel like that's not going to print well unless the direction of the line being printed is correct, ie coming out perpendicular to the previous layer across a void to another wall will bridge, but not a parallel (along the existing wall) line with so little overlap. 

If that's the desired functionality that part could be fixed by documentation change specifying that last bracket as 75..87 rather than 75..99.

igiannakas, 2024-07-14:
The slow down happens on this segment of code here:

https://github.com/SoftFever/OrcaSlicer/blob/854ce80a342e2e9460b6060b438c0857a2be4f3c/src/libslic3r/GCode/ExtrusionProcessor.hpp#L363

it checks the line offset in mm and if it’s less than the first entry it uses the original speed, if it’s larger than the last one it uses the final speed and anything in between it interpolates between the previous and current depending on the % difference.

the speed entries are passed here: 
If using curled perimeter slow down: 
https://github.com/SoftFever/OrcaSlicer/blob/854ce80a342e2e9460b6060b438c0857a2be4f3c/src/libslic3r/GCode.cpp#L5245

And if not using it here: 
https://github.com/SoftFever/OrcaSlicer/blob/854ce80a342e2e9460b6060b438c0857a2be4f3c/src/libslic3r/GCode.cpp#L5268

the overlap levels are defined here: https://github.com/SoftFever/OrcaSlicer/blob/854ce80a342e2e9460b6060b438c0857a2be4f3c/src/libslic3r/GCode.cpp#L5243

if not using curled overhang slow down anything with less than 13 % overlap will use the bridge speed.

igiannakas, 2024-07-14:
When I was porting over the curled perimeters implementation from prusa we did discuss this with Softfever. The idea why this is coded like this is that with under 13% overhang you are really printing bridges so the bridge speed is used and that yielded better results in SF’s experiments. 

However if you do want to slow down all the way and ignore the bridge speed, then the curled perimeters implementation does so already.

evilDave, 2024-07-14:
Thanks for the source links, I'll look there to see if I can see why it seems to not interpolate at the bottom, though that's probably a non issue as I doubt anyone needs to slow down for 10%

I'll look at the code but I think it's doing something odd right before a bridge. Printing a wall that overhangs more and more up to about 80% will slow down more and more, until 2 layers below the bridge height, and there it will do 2 layers at normal speed.

igiannakas, 2024-07-14:
It’s using the bridge speed. It may appear that it uses the external speed as in your profile your external speed is the same as the bridge speed. Try setting your external perimeter speed to say 200mm/sec or the bridge speed to 50mm/sec to better illustrate the changes in speed.

evilDave, 2024-07-14:
I don't think so. I had already done what you suggested so I could confirm it wasn't bridge speed. Regardless. It wouldn't be right to use the bridge speed on the 2 layers (at say 80% overhand) rather than a slower overhang speed would it?

Sorry I'm not at the computer right now, I'll post a picture to explain what I mean.

igiannakas, 2024-07-14:
It would help posting the STL too so I can slice it locally

evilDave, 2024-07-15:
So in making some screenshots of the last issue we were discussing, I now think that it's working as you say, but I have possibly uncovered a different issue. Some images to explain, and I'll add the smf and stl files too.

I was trying to test the slow down behaviour of overhang range 0%-99% as I thought that it would only bridge over 99%. That gives this large number of layers at bridge speed:

[image]

(Note also the lack of interpolation under 25% overhang)

From your explanation I wanted to re-test up to below 88% only, so created the next model. However I dragged it under the bed to save on filament (as I was printing these too to try to fix the actual issue that i was investigating which seemed to be extrusion issues at the point of 'large' flow changes) and only print the top. Slicing with part of the model below the bed gives:

[image]

As you can see there are 2 layers that I would think should be slowed, but they are actually printed at speed. This I think is a different issue though, as now that I just put the 87% model on the bed is produces:

[image]

Which is exactly what I was expecting from your explanation.

I don't know, do you want me to open a different ticket for the top layers when the object is below the bed?
My remaining thoughts on this ticket then are: 1. It's not clear and obvious from the documentation what exactly are the ranges of overhang for the slowdown. 2. There is an interpolation issue for under 25% overhang.

[Archive.zip](https://github.com/user-attachments/files/16230935/Archive.zip)

igiannakas, 2024-07-15:
Can you try downloading this build here:

I’ve made some fixes to the overhang segmentation granularity that may help with this. 

Build files are at the end of the page.

https://github.com/SoftFever/OrcaSlicer/actions/runs/9930756055

evilDave, 2024-07-15:
It looks much the same in terms of the layers at the top under the bridge, and the overhang under 25%

I see in the gcode there are some small changes, it seems like a few outer wall segments are merged together? like i see things like:
```
G1 X82.945 Y100.8 E.02233
G1 X112.01 Y100.8 E.8112
G1 X112.81 Y100.8 E.02233
```
becoming:
```
G1 X112.81 Y100.8 E.85585
```
which is probably an improvement anyway.
Was there something in particular I should look for?

igiannakas, 2024-07-15:
I thought what you’re seeing may have been an artefact of the line splitting not happening at the right place, which was fixed (together with not doing line splitting if not necessary) in this PR.

I’ll take a look at your attached project file tonight/tomorrow

TechnoSwiss, 2024-08-30:
Related to this video from Robert Cowan? https://www.youtube.com/watch?v=HcSOz-Lsxgg

closed 2024-12-06 by github-actions[bot]