OrcaSlicer/OrcaSlicer #4834: Error Send Model to FlashForge Adventurer 5M

pacocatech, 2024-04-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

FlashForge Adventurer 5M

### How to reproduce

1 Configure network connection 
2 Slice Model
3 Print
4 Error on send model to printer
[image]
[image]
[image]
[image]

### Actual results

Error on send model to printer
[image]

### Expected results

Send model to printer with success

### Project file & Debug log uploads

[crash_Tue_Mar_26_23_29_08_0.log](https://github.com/SoftFever/OrcaSlicer/files/14823374/crash_Tue_Mar_26_23_29_08_0.log)
[bust.zip](https://github.com/SoftFever/OrcaSlicer/files/14823382/bust.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

pacocatech, 2024-04-01:
When I make a smaller file, for example, the standard orca cube, the file is sent successfully, it must have something to do with the size of the file, in my example the file is more than 100mb

pacocatech, 2024-04-01:
@FlashforgeOfficial can u help?

savacartonner, 2024-04-02:
I have the same problem with my X1C from Bambu Lab

FlashforgeOfficial, 2024-04-07:
> @FlashforgeOfficial can u help?

@iherbak Hi could you help take a look on this one?

tlochi, 2024-04-11:
Same problem on my Adventure 5M. A consistent amount of time passes during uploading a file before Orca decides there is an error. On the printer interface, the upload status bar continues to fill up normally until Orca generates the error message.

iherbak, 2024-04-11:
I’ve created a windows command line tool to provide something to experiment with. https://github.com/iherbak/OrcaSlicer/releases/download/Latest/5MWifi.zip

Could you try to use it for uploading the gcode? You could change the timeout and the size in bytes to split the file on sending. I was unable to reproduce the issue myself. I hope we could find the issue so i could adapt the code.

preacher4gh, 2024-05-02:
I have an Adventure 3 and had the same issue.
Command line tool worked prerfect for me, but surprisingly, after looking at the machine, the test file and even those that timed out on orca slicer were there, but without preview image.
So I wonder, which Gcode flavor and which image format for preview to choose.

thhdragon, 2024-06-08:
> I’ve created a windows command line tool to provide something to experiment with. https://github.com/iherbak/OrcaSlicer/releases/download/Latest/5MWifi.zip
>
> Could you try to use it for uploading the [...]

would you mind helping me out with that? what are the defaults for the port and the [data_wait_timeout] [data_packet_size]? not sure what values to start with.

iherbak, 2024-06-08:
Hi. The dafult port is 8899 waittime try 60 first packet size in bytes so
try 10240.

thhdragon ***@***.***> (időpont: 2024. jún. 8., Szo, 6:48)
ezt írta:

Supremeb1, 2024-06-11:
Adventurer 5m same thing happens to me with 6mb file.

makc9, 2024-07-20:
> I’ve created a windows command line tool to provide something to experiment with. https://github.com/iherbak/OrcaSlicer/releases/download/Latest/5MWifi.zip
>
> Could you try to use it for uploading the [...]

thank you! with long timeout downloads well. how to change timeout in orca?

goldfish408, 2024-07-25:
Same issue for me with a FlashForge Adventurer 5M Pro with Windows 11.

xlordxcheater, 2024-08-29:
Same happening here with a Adventurer 5m with files over 12mb. Sending the files via Wifi

xlordxcheater, 2024-10-26:
> I’ve created a windows command line tool to provide something to experiment with. https://github.com/iherbak/OrcaSlicer/releases/download/Latest/5MWifi.zip
>
> Could you try to use it for uploading the [...]

How can I use the command line tool? I only got a zip file with a 5MWifi.exe and other files

Hugofer27, 2024-11-15:
> > @FlashforgeOfficial can u help?
> 
> @iherbak Hi could you help take a look on this one?

Fix your configuration on Orca Slicer

xlordxcheater, 2024-11-15:
> > > @FlashforgeOfficial can u help?
> > 
> > 
> > @iherbak Hi could you help take a look on this one?
> 
> Fix your configuration on Orca Slicer

Can you please Elaborate on this?

iherbak, 2024-11-15:
> > > > @FlashforgeOfficial can u help?
> > > 
> > > 
> > > @iherbak Hi could you help take a look on this one?
> > 
> > 
> > Fix your configuration on Orca Slicer
> 
> Can you please Elaborate on this?

I guess the issue is with the serial connection instability in case of uploading bigger files.
Since Orca-Flashforge is available i found they added an more reliable rest communication to the firmware. I made an adaptation.
https://github.com/iherbak/OrcaSlicer/releases/tag/Latest

Could you please try this version. The instruction on how to setup the connection is written in the notes.

rivens623, 2024-11-21:
@iherbak - I installed the latest and a smaller STL file (12 MB) and it worked. I will try a bigger project file I am working on after this one is done.  Thanks!!

rivens623, 2024-11-22:
@iherbak - Just tried to send a bigger file (164MB) and it failed to send.  "Error uploading to print host: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond."  The printer shows that it is receiving a file when I try to send it.

rivens623, 2024-11-22:
@iherbak do you have any specific filters you would want applied for wireshark or just start the monitoring, attempt to send the file to the printer and then send the results?

iherbak, 2024-11-22:
@rivens623 no filter neccessary. Only the recorded file. Start it before hitting upload and stop it after the error happens in orca. I hope we could see something. Thanks

rivens623, 2024-11-23:
[WireSharkResults.zip](https://github.com/user-attachments/files/17883187/WireSharkResults.zip)

@iherbak - Here are the wireshark results.  Two files, one trying to send the file w/o printing, the other with printing command via OrcaSlicer.

iherbak, 2024-11-24:
@rivens623 Examined the results. You are using the previous version which is using serial communication. (I see the gcode commands). I tried to reproduce the issue with no luck. 
However since Orca-flashforge is out a new API connection is available with the latest firmwares that is more reliable. 
Could you please try the latest version at my github. https://github.com/iherbak/OrcaSlicer/releases/tag/Latest. The instructions on how to configure is on the page.

I am waiting on feedbacks before creating a pull request to the Softfever repo.

rivens623, 2024-11-24:
<FACEPALM> @iherbak - That did it.  Sorry I thought I had the right one installed.  The version numbers were the same for the correct one I installed from the previous.

rfkarpinski, 2024-12-03:
> @rivens623 Examined the results. You are using the previous version which is using serial communication. (I see the gcode commands). I tried to reproduce the issue with no luck. However since [...]

Hello!
Works like a charm. I have downloaded 2.3.0-dev, configure SN and printer ID and 62MB works.
Thanks a lot!

esistgut, 2025-03-06:
@iherbak hey, did you push a merge request on mainline orca?

istvanherbak, 2025-03-06:
@esistgut Hi, i had a merge request, but they commented on several time and lamented if we should preserve the old serial connection as well, so i gave up on that. Currently i added the dialog clearing before print, and i will prepare a new merge request including that as well. Till then in my own branch you could try it.

closed 2025-06-12 by github-actions[bot]