OrcaSlicer/OrcaSlicer #4848: Support for other software profiles import

AzazelHD, 2024-04-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hello. I have an Ender 3 V3 SE and I have spent some time trying to tune it to make prints look as best as possible, but the results are always worse than when generating the gcode with Creality Print (creality official slicer).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Could you add a feature to orca can import the profiles from other slicers please?

cochcoder, 2024-04-03:
Not related to this issue specifically, even though it would be a nice feature, you could create another issue listing the differences between the two profiles and what should be changed. If you give me the link to this new issue I'll work on it.

AzazelHD, 2024-04-04:
Sorry @cochcoder is this for me? Because i am not sure. Many things are similar but with a different naming, also some other features arent on both softwares. I could share the default Creality Print default (0.2 mm) and high quality (0.1 mm) profiles
Hope they are useful
[Creality_0.1mm.txt](https://github.com/SoftFever/OrcaSlicer/files/14873258/Creality_0.1mm.txt)
[Creality_0.2mm.txt](https://github.com/SoftFever/OrcaSlicer/files/14873259/Creality_0.2mm.txt)

cochcoder, 2024-04-04:
> Sorry @cochcoder is this for me? Because i am not sure. Many things are similar but with a different naming, also some other features arent on both softwares. I could share the default Creality [...]

Thanks for the profiles, I'll look into to fixing the issues you mentioned

AzazelHD, 2024-04-04:
I had to change the extension cuz github didnt allow me to upload it otherwise. Original extension is .cxprofile

AzazelHD, 2024-05-05:
Omg, sorry for such a late response. I was trying to manually copy the settings from Creality Print into Orca, asking for some help on 3d printing discords, because naming wasnt the same. So probably it was my bad, cuz i wasnt putting the correct values on the correct fields.

cochcoder, 2024-05-05:
There should be a default setting for the sender 3 V3 SE, does that work well?

closed 2024-08-11 by github-actions[bot]