OrcaSlicer/OrcaSlicer #4849: Printer profile for Artillery Sidewinder X4 PRO

bednarekpiotr, 2024-04-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I cannot find profile for this printer

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

I would like to have profile for Artillery Sidewinder X4 PRO

Cookie-fighter, 2024-04-03:
Same goes for the X4 plus. Of course they are very new.
But clearly it would be benefical to have a base profile.

hybotics, 2024-04-14:
Please add the X4 Pro and X4 Plus profiles.

dan4ik3d, 2024-04-30:
**I NEED IT cz Cura much better than @prusia**

closed 2024-08-07 by github-actions[bot]