OrcaSlicer/OrcaSlicer #4853: Scarf seam: a G-Code path goes beyond the max print height

superfloh247, 2024-04-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

c663f43

### Operating System (OS)

macOS

### OS Version

14.3.1

### Additional system information

apple silicon arm64

### Printer

Bambu P1P

### How to reproduce

slice example file

### Actual results

Scarf seam produces incorrect G-Code

### Expected results

correct G-Code

### Project file & Debug log uploads

[Scraf-seam-error.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14849617/Scraf-seam-error.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

superfloh247, 2024-04-03:
might be related to #4754

vgdh, 2024-04-03:
try PR https://github.com/SoftFever/OrcaSlicer/pull/4843

superfloh247, 2024-04-03:
#4843 seems to fix this

closed 2024-04-03 by superfloh247