OrcaSlicer/OrcaSlicer #4859: Outer wall wiping removed by elephant foot compensation + other settings

czlr, 2024-04-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Education

### Printer

Bambu P1S

### How to reproduce

1. Start from the attached project file
2. Slice with [Elephant foot compensation] enabled. E.g., 0.15mm to see the issue

### Actual results

On layer 1, the outermost wall isn't ended with a wipe operation. Note: "wipe" in this case means the wiping associated with retractions, via [Wipe while retracting].

[image]

**Other things I noticed:**
1. The issue might be from a combination of [Avoid crossing walls]=ON and [Elephant foot compensation]>0. Since the issue doesn't happen with [Avoid crossing walls]=OFF and [Elephant foot compensation]=0.
2. As mentioned, setting [Elephant foot compensation]=0 fixes the issue. However, then setting [Seam position]=Nearest (previously "Aligned") seems to break it again. This shouldn't be related to Extruder [Travel distance threshold], since the z-hop still gets generated. In this state, setting [Arc fitting]=OFF (previously "ON") fixes the issue again. So there are some weird interactions here...
3. Not sure if related: "Inner/Outer/Inner" wall order doesn't generate correctly on Layer 1 with 2 walls. My attached settings have [Only one wall on first layer]=ON, but if you disable that, you'll see that "Inner/Outer/Inner" prints the inner wall before the outer wall. (Expected: with 2 walls, it should be printing the outer wall before inner wall, like "Outer/Inner") So at the very least, it seems to have special logic for layer 1.

### Expected results

Layer 1 outer walls should be generated with wipes regardless of most other settings: [Elephant foot compensation], etc. In my mind, perhaps the only reason it'd be disabled would be due to the subsequent move falls below the retraction [Travel distance threshold]. But whenever a z-hop + retraction is generated, I'd expect there to always be a wipe (when configured) too.

### Project file & Debug log uploads

[elephant_foot_wiping_bug.zip](https://github.com/SoftFever/OrcaSlicer/files/14857171/elephant_foot_wiping_bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

frech-tech, 2024-04-07:
Hey,
I have the same issue. With "elephant foot compensation" + "Avoid crossing walls" the wipe movment is missing. This leads to strange blobs near the seam. 
"elephant foot compensation = 0" + "Avoid crossing walls : Off" 
[image]
"elephant foot compensation = 0.16" + "Avoid crossing walls: On" 
[image]

will-flaugher, 2024-04-09:
Seems potentially related to the issue in #4226, in case it helps to either consolidate work or have more data to review.

CDFER, 2024-06-01:
Just to add few datapoints to help debug this I have basically the same issue with the 2.1.0 beta, (but I had the same problem in 2.0.0 too).  

### Simple Cube Test
#### Wipe working correctly with avoid crossing walls off:
[image]

#### Wipe missing with avoid crossing walls on:
[image]

Here is the test file:
[Cube_0.2mm_PETG_MINI_8m52s.zip](https://github.com/user-attachments/files/15523394/Cube_0.2mm_PETG_MINI_8m52s.zip)

### Shape Test
### Wipe works correctly with a cube but doesn't work with circular objects (10% infill)
[image]

### Wipe works on all objects (0% infill)
[image]

Anyways this is beyond my programming skills but hopefully it helps you guys figure it out. I've really been impressed with OrcaSlicer, thanks for the amazing work you all have done :)

closed 2024-09-07 by github-actions[bot]